OpenHantek/openhantek #8: Store firmware files in repository, wait for hantek response

davidgraeff, 2015-10-11:
The mail content send to  support@hantek.com.cn:

> Dear Sir or Madam,
> 
> I kindly ask you for permission to store the firmware of some of your older devices
> "DSO-2090", "DSO-2150", "DSO-2250", "DSO-5200", "DSO-5200A" but also one
> of your current ones ("DSO-6022BL/BE") in a public accessible way (say for example github.com).
> 
> This would reduce maintainance costs and improve usability of our open source
> software "OpenHantek" (https://github.com/OpenHantek/openhantek, http://openhantek.org/),
> which exists because your original software does not run within Linux and MacOSX.
> The software is GPLv3 licensed and therefore will not be used for commercial purposes.
> 
> We would love to make the software compatible to even more of your DSOs and it would
> be great to have your permission for storing all necessary binary data for also all in the future
> supported DSO models in our git repository.
> 
> If you are interested in a faster development speed for this software, that allows Linux and MacOSX
> users to be (non-official) customers of yours, you may think about providing us access to
> usb protocol documentation. But this is not required.
> 
> Thanks,
> David Graeff

davidgraeff, 2015-10-11:
If we are allowed to store firmware files, I would propose to move the firmware-extract tool to a separate repository, remove it from here and call it "deprecated". We would then store firmware for all supported DSOs right here in the repo. Any opinions?

OliverHaag, 2015-10-11:
Wouldn't call it deprecated, but move it to another repository (Would like to do that anyway...) and just not reference it in the installation instructions anymore (Since the tool is still required, at least for us).
Had contact to one of the Hantek devs once and he was nice and helpful. Let us know how it's going :)

davidgraeff, 2015-10-15:
I'm not able to contact Hantek via their mail addresses (propably a problem with the Great Firewall).

>    ----- Transcript of session follows -----
> ... while talking to mailmx2.dns.com.cn.:
> >>> DATA
> <<< 451 Requested mail action not taken: mailbox unavailable 
> <service@hantek.com.cn>... Deferred: 451 Requested mail action not taken: mailbox unavailable 
> <<< 451 Requested mail action not taken: mailbox unavailable 
> <support@hantek.com.cn>... Deferred: 451 Requested mail action not taken: mailbox unavailable 
> <<< 503 Must enter RCPT command first 
> Warning: message still undelivered after 4 hours
> Will keep trying until message is 5 days old

OliverHaag, 2015-10-16:
I've mailed the dev I had contact with, maybe he's still working on the project and can help with this :)

davidgraeff, 2016-02-05:
As we do not get a response, I assume they don't care. I propose to upload all binaries we get our hands on and dismiss the old binary extractor tool.

PR #21 (2016-03-02): Commit hantek firmware
The script to download the Hantek drivers was broken because Hantek's website moved the download location, changed from rar files to zip files, etc.

This avoids similar problems in the future, and it should make installation much easier for users too. I've also fixed some inconsistencies between the name the `extractfw` utility gives the firmware files, and the filenames the udev rule expects.

Note that I've only tested the 2090 firmware, not the other models from the DSO2000 series. Those models were listed in the original `fwget.sh` script so I've added their firmwar [... cut]
closed 2016-03-02 by davidgraeff