OpenHantek/openhantek #10: Broken GUI layout

jflesch, 2015-11-10:
Hello,

I just tried installing Openhantek on GNU/Linux Debian. It seems there is a major problem with the layouts of the elements of the GUI. It makes a lot of features unusable.

[image]

GUI elements can be rearranged by the user, but when restarting OpenHantek, everything is a mess again.

Distribution : GNU/Linux Debian Jessie (8.2), 64bits
WM: Gnome 3 (dual screen)

geissdoerfer, 2015-11-21:
Same configuration and same issue for me. Apparently there's some buggy threading. If i try to change some settings (AC/DC):
```QObject: Cannot create children for a parent that is in a different thread.
(Parent is Hantek::Control(0x1c37960), parent's thread is QThread(0x1a2fdb0), current thread is Hantek::Control(0x1c37960)```

davidgraeff, 2015-11-21:
The current GUI is written in pure c++ and therefore not very maintenance friendly. I'm working on a successor written in a declarative style and with the current technology (Qt QML) in the other branch but still on holiday at the moment :)

OliverHaag, 2015-11-26:
The threading issue and the GUI issue seem to be unrelated on first sight to me. I agree with David though - best wait for the QML rewrite and recheck both issues then.
Back in the Qt 4 times pure C++ was the best way to do it imo but nowadays I would also use QML since it's just much easier maintainable. Most probably some things have changed with Qt 5 (Guess Qt 5 was used, but this wasn't specified in the reports).

geissdoerfer, 2015-11-26:
Ah okay i thought it was related, as for me the functionality is also affected and it kind of looked like the GUI elements could not communicate with control parts. Haven't had a closer look though tbh. Also i'm not sure about which Qt i was actually using, will check and try both. Anyway @davidgraeff, had a look into your branch and it looks very promising :+1: The killer feature for me would be a clean programmatic interface, that i can wrap for matlab/python

cskau, 2015-12-10:
I ran into the same issue a couple days ago, and I went in an created a fix for at least part of the issue in this PR:
https://github.com/OpenHantek/openhantek/pull/11

I do not think this is related to threading.

davidgraeff, 2016-03-03:
In my setup this problem is fixed, please open another issue if it is not for you.

closed 2016-03-03 by davidgraeff