OpenHantek/openhantek #12: Problems building extractfw what gives?!

jangles007, 2015-12-18:
I can't build the extract tool for the life of me:
[root@Jeff-PC openhantek-extractfw]# ./configure
bash: ./configure: No such file or directory

root@Jeff-PC openhantek-extractfw]# make
make: *** No targets specified and no makefile found.  Stop.

I followed the instructions exactly, what am I doing wrong?

Also, the fwget tool is bunged up as the BASEURL has moved locations. The tool does not work.

Can someone please help me?

gipi, 2015-12-18:
You have to generate using the autotools the ``Makefile``: in a Debian system the following instructions work for me

```
# apt-get install libiberty-dev binutils-dev
$ cd openhantek-extractfw
$ autoreconf -vi
$ ./configure
$ make
  CC       extractfw.o
  CCLD     openhantek-extractfw
```

jangles007, 2015-12-18:
Hello, I am running Fedora 23. I don't your instruction will work for me 
butit maybe a clue?

Thanks for your help!

Jeff

-- 
Regards,

Jeff Doucette
Electronics Technician
Hi-Tronics Repair & Restoration
250 City Centre Ave. Bay 236 Upper Level
Ottawa, Ontario
K1R-6R7
Phone: 613-725-1157
Web: www.hi-tronics.ca
Facebook: www.facebook.com/ottawaelectronicsrepair

jangles007, 2015-12-19:
Now when I installed autoconf I can run configure but I get this error now:

[root@Jeff-PC openhantek-extractfw]# ls
90-hantek.rules.in  config.log     COPYING      INSTALL      NEWS
AUTHORS             config.status  depcomp      install-sh   README
autom4te.cache      configure      extractfw.c  Makefile.am
ChangeLog           configure.in   fwget.sh     missing
[root@Jeff-PC openhantek-extractfw]# ./configure
./configure: line 2036: AM_INIT_AUTOMAKE: command not found
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for dlopen in -ldl... yes
checking for deflate in -lz... yes
checking for objalloc_create in -liberty... yes
checking for bfd_openr in -lbfd... yes
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking malloc.h usability... yes
checking malloc.h presence... yes
checking for malloc.h... yes
checking for string.h... (cached) yes
checking for stdlib.h... (cached) yes
checking for GNU libc compatible malloc... yes
configure: creating ./config.status
config.status: error: cannot find input file: `Makefile.in'

OliverHaag, 2015-12-21:
Try running aclocal first. The generated configure script is also contained in the extractfw zip on http://sourceforge.net/projects/openhantek/files/, so if you get it there you won't have to do this.

But afaik fwget won't work anymore because the URLs to the driver files have changed.

jangles007, 2015-12-21:
Hi Oliver,

Thank you for helping me.

I did as you instructed however the output is as follows:

[root@Jeff-PC openhantek-extractfw]# aclocal
aclocal: warning: autoconf input should be named 'configure.ac', not 
'configure.in'

Please advise,

Jeff

OliverHaag, 2015-12-21:
That's just a warning, but the filename should be changed on occasion. The docs also need an update, or we just put the generated configure script on github too.
@davidgraeff Any plans on switching this to CMake too? May be an overkill but it would be consistent with the main project.

jangles007, 2015-12-21:
Ok, so I ran configure afterwards now I get more error:

[root@Jeff-PC openhantek-extractfw]# ./configure
checking whether make supports nested variables... yes
checking for a BSD-compatible install... /bin/install -c
checking whether build environment is sane... yes
/home/jeff/Downloads/OpenHantek-openhantek-1c1c21f/openhantek-extractfw/missing: 
Unknown `--is-lightweight' option
Try 
`/home/jeff/Downloads/OpenHantek-openhantek-1c1c21f/openhantek-extractfw/missing 
--help' for more information
configure: WARNING: 'missing' script is too old or missing
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking for dlopen in -ldl... yes
checking for deflate in -lz... yes
checking for objalloc_create in -liberty... yes
checking for bfd_openr in -lbfd... yes
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking malloc.h usability... yes
checking malloc.h presence... yes
checking for malloc.h... yes
checking for string.h... (cached) yes
checking for stdlib.h... (cached) yes
checking for GNU libc compatible malloc... yes
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: error: cannot find input file: `Makefile.in'

jangles007, 2016-01-07:
I'm stuck! All I see is Makefile.am.

davidgraeff, 2016-02-05:
If you actually upload binaries to the repo, openhantek-extractfw would be obsolet. But yes I'll port that over, too.

3esmit, 2016-02-29:
run automake --add-missing

closed 2016-03-02 by davidgraeff

kyberkiller, 2017-02-06:
fwget.sh says:

...
BFD: tmp/Dso2090AMD641.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .text
BFD: tmp/Dso2090AMD641.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .rdata
BFD: tmp/Dso2090AMD641.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .data
BFD: tmp/Dso2090AMD641.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .pdata
Section .data found (starting at 0x1e80, 9784 bytes)
Symbol _firmware found (offset 0xffffffffffffe180, 0 bytes)
Symbol _loader found (offset 0xffffffffffffe180, 9784 bytes)

BFD: tmp/Dso5200Ax862.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .text
Section .data not found
bfd_check_format: File format not recognized
bfd_check_format: File format not recognized
BFD: tmp/Dso5200x861.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .text
BFD: tmp/Dso5200x861.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .data
Section .data found (starting at 0x08a0, 9616 bytes)
Symbol _firmware found (offset 0x0000, 8296 bytes)
Symbol _loader found (offset 0x2068, 1320 bytes)
Saving firmware as tmp/dso5200x86-firmware.hex
Saving loader as tmp/dso5200x86-loader.hex
BFD: tmp/Dso5200x862.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .text
Section .data not found
.
.
:
.

...
--blaa blaa blaa...

.
.
.
.
(6022 zip added to same folder as others..)

BFD: tmp/Hantek6022BEAMD641.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .text
BFD: tmp/Hantek6022BEAMD641.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .rdata
BFD: tmp/Hantek6022BEAMD641.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .data
BFD: tmp/Hantek6022BEAMD641.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .pdata
Section .data found (starting at 0x1180, 10144 bytes)
Symbol _firmware found (offset 0xffffffffffffee80, 0 bytes)
Symbol _loader found (offset 0xffffffffffffee80, 10144 bytes)
Can't get firmware contents: Bad value
BFD: tmp/Hantek6022BEAMD642.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .text
BFD: tmp/Hantek6022BEAMD642.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .rdata
BFD: tmp/Hantek6022BEAMD642.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .data
BFD: tmp/Hantek6022BEAMD642.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .pdata
Section .data found (starting at 0x3500, 272 bytes)
Symbol _firmware found (offset 0xffffffffffffcb00, 0 bytes)
Symbol _loader found (offset 0xffffffffffffcb00, 272 bytes)
Can't get firmware contents: Bad value
bfd_check_format: File format not recognized
bfd_check_format: File format not recognized
BFD: tmp/Hantek6022BEX861.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .text
BFD: tmp/Hantek6022BEX861.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .rdata
BFD: tmp/Hantek6022BEX861.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .data
Section .data found (starting at 0x0c00, 9864 bytes)
Symbol _firmware found (offset 0xfffffffffffff400, 0 bytes)
Symbol _loader found (offset 0xfffffffffffff400, 9864 bytes)
Can't get firmware contents: Bad value
BFD: tmp/Hantek6022BEX862.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .text
BFD: tmp/Hantek6022BEX862.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .rdata
BFD: tmp/Hantek6022BEX862.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .data
Section .data found (starting at 0x2780, 8 bytes)
Symbol _firmware found (offset 0xffffffffffffd880, 0 bytes)
Symbol _loader found (offset 0xffffffffffffd880, 8 bytes)
Can't get firmware contents: Bad value
....

PR #21 (2016-03-02): Commit hantek firmware
The script to download the Hantek drivers was broken because Hantek's website moved the download location, changed from rar files to zip files, etc.

This avoids similar problems in the future, and it should make installation much easier for users too. I've also fixed some inconsistencies between the name the `extractfw` utility gives the firmware files, and the filenames the udev rule expects.

Note that I've only tested the 2090 firmware, not the other models from the DSO2000 series. Those models were listed in the original `fwget.sh` script so I've added their firmwar [... cut]