OpenHantek/openhantek #13: Error while build branch `mr_split_gui_backend`

giomasce, 2016-01-03:
While building branch `mr_split_gui_backend` with `cmake` I obtain:

    $ cmake ..
    CMake Error at openhantek2/CMakeLists.txt:3 (add_subdirectory):
      The source directory
    
        /home/giovanni/progetti/hantek/openhantek/openhantek2/plot
    
      does not contain a CMakeLists.txt file.
    
    
    -- Configuring incomplete, errors occurred!
    See also "/home/giovanni/progetti/hantek/openhantek/build/CMakeFiles/CMakeOutput.log".
    See also "/home/giovanni/progetti/hantek/openhantek/build/CMakeFiles/CMakeError.log".

Maybe @davidgraeff forgot to add a file?

giomasce, 2016-01-03:
Also, file `openhantek2/src/currentdevice.h` refers `plot/qpscaleengine.h`, while my `plot` subdirectory is empty.

davidgraeff, 2016-02-05:
@giomasce The branch is not complete in any way. I have to discuss some changes with @OliverHaag first and desperatly need some usb sniffed packages from the supported DSOs to test the newly created libs.

giomasce, 2016-02-05:
I will be able to provide you with sniffs from 6022BE during the weekend, if they are useful.

davidgraeff, 2016-02-05:
@giomasce Thank you for this offer. I do own a 6022 myself, it's more a matter with the older devices :)