OpenHantek/openhantek #15: softDSP SDS 200A

DinBan, 2016-01-18:
Hello, let me know please, will this software(openhantek) works with USB oscilloscope softDSP SDS 200A ?
http://www.softdsp.com/product/sds200a_01.htm
Also is there any manual to install this software on OS X
Thank you in advance!

OliverHaag, 2016-01-18:
Nope, not yet.

Supporting other manufacturers scopes is basically possible and should get easier with the rework done by @davidgraeff at the moment. So in case you have C++ experience and some time feel free to start working on this.

This is an issue tracker btw., not a forum. But since supporting this scope familiy might make sense in the future (as long as someone who has access to such a device starts working on it) I'll leave this issue open.

DinBan, 2016-01-18:
Thank you for early replay! I have this device and ready to test software if someone can modify it for SDS 200A, but I'm don't have so deep experience in C++ to do it by my self :(

davidgraeff, 2016-02-05:
I'll look into it. But the base protocol implementation should be done by someone who owns that device, but maybe the protocol is similar to one of the supported devices.

closed 2017-12-18 by davidgraeff