OpenHantek/openhantek #18: Does not work with DSO-2250

usern1983, 2016-02-14:
$ openhantek
Warning: This Hantek DSO model isn't supported officially, so it may not be working as expected. Reports about your experiences are very welcome though (Please open a feature request in the tracker at https://sf.net/projects/openhantek/ or email me directly to oliver.haag@gmail.com). If it's working perfectly I can remove this warning, if not it should be possible to get it working with your help soon.
Getting capture state failed: Operation timed out
Getting capture state failed: Operation timed out

=======my_OS: http://www.calculate-linux.ru/==================================
$ uname -a 
Linux usern 4.4.1-calculate #1 SMP PREEMPT Fri Feb 12 15:13:56 MSK 2016 x86_64 AMD A4-5000 APU with Radeon(TM) HD Graphics AuthenticAMD GNU/Linux

я изменил файл udev:
# cat /etc/udev/rules.d/90-hantek-2250.rules 

# Hantek DSO-2250
#SUBSYSTEM=="usb", ACTION=="add", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="4b4/2250/*", RUN+="/sbin/fxload -t fx2 -I @hantekfirmwaredir@/dso2250-firmware.hex -s @hantekfirmwaredir@/dso2250-loader.hex -D $env{DEVNAME}"
#SYSFS{idVendor}=="04b5", SYSFS{idProduct}=="2250", MODE="0660", GROUP="plugdev"

SUBSYSTEM=="usb", ACTION=="add", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="4b4/2250/*", RUN+="/sbin/fxload -t fx2 -I /usr/local/share/hantekdso/DSO2250_firmware.hex -s /usr/local/share/hantekdso/DSO2250_loader.hex -D $env{DEVNAME}"
ATTRS{idVendor}=="04b5", ATTRS{idProduct}=="2250", MODE="0660", GROUP="plugdev"

# lsusb 
Bus 002 Device 002: ID 0bda:5775 Realtek Semiconductor Corp. HP "Truevision HD" laptop camera
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 003: ID 04b5:2250 ROHM LSI Systems USA, LLC 
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

# ls -l /dev/bus/usb/005/003 
crw-rw---- 1 root plugdev 189, 514 фев 14 12:58 /dev/bus/usb/005/003

=========without udev rule for 2250========================================
# lsusb
Bus 002 Device 002: ID 0bda:5775 Realtek Semiconductor Corp. HP "Truevision HD" laptop camera
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 004: ID 04b4:2250 Cypress Semiconductor Corp. 
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

# ls -l /dev/bus/usb/005/004 
crw-rw-r-- 1 root usb 189, 515 фев 14 14:47 /dev/bus/usb/005/004
========================================================================

http://www.hantek.com.cn/down.aspx?url=http%3a%2f%2fwww.hantek.com%2fProduct%2fDSO2000%2fDSO2250_Driver.zip

# unzip -l DSO2250_Driver.zip 
Archive:  DSO2250_Driver.zip
  Length      Date    Time    Name
---------  ---------- -----   ----
        0  12-18-2014 09:21   DSO2250_Driver/
    27464  09-25-2014 09:02   DSO2250_Driver/DSO2250AMD641.sys
    45768  09-25-2014 09:02   DSO2250_Driver/DSO2250AMD642.sys
    41288  09-25-2014 09:02   DSO2250_Driver/DSO2250IA641.sys
   123720  09-25-2014 09:02   DSO2250_Driver/DSO2250IA642.sys
    10250  09-25-2014 09:02   DSO2250_Driver/dso2250usb.cat
     5188  09-25-2014 09:00   DSO2250_Driver/dso2250usb.inf
    23888  09-25-2014 09:02   DSO2250_Driver/Dso2250x861.sys
    25544  09-25-2014 09:02   DSO2250_Driver/Dso2250x862.sys
---------                     -------
   303110                     9 files

=====================================================================

~/openhantek/openhantek-extractfw $ ./openhantek-extractfw ../DSO2250_Driver/DSO2250AMD641.sys 
BFD: ../DSO2250_Driver/DSO2250AMD641.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .text
BFD: ../DSO2250_Driver/DSO2250AMD641.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .rdata
BFD: ../DSO2250_Driver/DSO2250AMD641.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .data
BFD: ../DSO2250_Driver/DSO2250AMD641.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .pdata
Section .data found (starting at 0x1e80, 9784 bytes)
Symbol _firmware found (offset 0xffffffffffffe180, 0 bytes)
Symbol _loader found (offset 0xffffffffffffe180, 9784 bytes)
Can't get firmware contents: Bad value

~/openhantek/openhantek-extractfw $ ./openhantek-extractfw ../DSO2250_Driver/DSO2250IA641.sys 
bfd_check_format: File format not recognized

~/openhantek/openhantek-extractfw $ ./openhantek-extractfw ../DSO2250_Driver/Dso2250x861.sys 
BFD: ../DSO2250_Driver/Dso2250x861.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .text
BFD: ../DSO2250_Driver/Dso2250x861.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .data
Section .data found (starting at 0x08a0, 9504 bytes)
Symbol _firmware found (offset 0x0000, 8184 bytes)
Symbol _loader found (offset 0x1ff8, 1320 bytes)
Saving firmware as ../DSO2250_Driver/dso2250x86-firmware.hex
Saving loader as ../DSO2250_Driver/dso2250x86-loader.hex

davidgraeff, 2016-02-17:
thanks for the detailed report. @OliverHaag knows that part better. But it seems the model is not detected right. The 2250 is actually supported. You also have usb read/write access as far as I can see (at least if SELINUX, AppArmor or posix ACL is not interfering our communication).

3esmit, 2016-02-29:
You should notice that dso udev rules points to a file named dso2550-firmware.hex not   dso2250x86_firmware.hex. Update 90-hantek.rules or rename you hex file. Also if you are extracting x86.sys for x86 systems, and use 64.sys (that is not supported by current extractfw) for 64 systems.

usern1983, 2016-03-01:
> You should notice that dso udev rules points to a file named dso2550-firmware.hex not dso2250x86_firmware.hex. Update 90-hantek.rules or rename you hex file.

It's clear [Это понятно].

> Also if you are extracting x86.sys for x86 systems, and use 64.sys (that is not supported by current extractfw) for 64 systems.

I thought so [ я так и думал].
is there any way to get the x64 firmware? [Есть ли способ получить x64 прошивку?]

3esmit, 2016-03-01:
I didn't dig into firmware extractor but probably there is a way to extract x64 firmware, the current software *simply* don't know how to do it. 
For now you can workaround this installing and compiling this program in x86 arch, that might get you some work (and a lot of libraries to install).

usern1983, 2016-03-01:
Now I use a vbox+win7x64. It is surprising that a guest I bring usb device after use udev (which firmware x86) and win-native program works great.
We must try chroot32.

[Сейчас я использую vbox+win7x64. Удивительно, что гостю я передаю usb устройство после работы udev (а это прошивка x86) и родная win-программа прекрасно работает.

Надо попробовать chroot32.]

closed 2017-12-17 by davidgraeff