OpenHantek/openhantek #19: Questions about Hantek DSO 2150

Mesqualito, 2016-02-21:
After downloading the newest Hantek-Driver from their website, which are Windows 8-compatible, I got a working firmware extracted with:
> cp Dso2150x861.sys DSO21501.SYS
and
> dsoextractfw

Openhantek 0.2.0 starts up with
> sudo openhantek
Warning: This Hantek DSO model isn't supported officially, so it may not be working as expected. Reports about your experiences are very welcome though (Please open a feature request in the tracker at https://sf.net/projects/openhantek/ or email me directly to oliver.haag@gmail.com). If it's working perfectly I can remove this warning, if not it should be possible to get it working with your help soon.
QObject: Cannot create children for a parent that is in a different thread.
(Parent is Hantek::Control(0x16dce70), parent's thread is QThread(0x16593b0), current thread is Hantek::Control(0x16dce70)

The program shows, that the DSO 2150 is connected in the status line:
"Device found: Hantek DSO-2150 (002:013)"

I can measure the test signal of 1 kHz shown in the screenshot. I wonder why there is this much "interference", no clear line display.
Is it normal, that I have to change settings (e.g. samplerate or timebase or voltage for CH1) to get an updated view? There is no "live-view", when I disconnect the probe from the test-signal-port.

[image]

Mesqualito, 2016-02-21:
O.k., after playing around with the new settings which aren't documented yet in your screenshots, I get the test-signal like your screenshot like this:

[image]

What is the Spektrum-Analysis - and why don't I have signals?

Thanks a lot!

Sincerely,
Jochen

jedi7, 2016-11-26:
Hi,
regarding the "inteference". It is quite normal also on normal oscilloscopes (analog&digital). 
Because the horizontal line is created from many measured points (voltage) and the measure is not always the same (noise, etc)
The vertical line is created normally only from "two" measured points, then it can be interpreted as long line.

davidgraeff, 2017-12-18:
I guess the question is resolved?

closed 2018-01-14 by davidgraeff