OpenHantek/openhantek #22: Integrate firmware loading

davidgraeff, 2016-03-02:
At the moment we need a special udev file which instructs an external program (fxload) to load the firmware to a DSO. The actual procedure and code is very simple and the uploading should be done by OpenHantek instead.

Reference: https://github.com/libusb/libusb/blob/master/examples/fxload.c (ezusb.h, ezusb.c are the interesting files that we may use directly, GPL licensed)

3esmit, 2016-03-02:
This enhancement will provide ARM (raspberry pi/raspbian) compatibility to OpenHantek?

davidgraeff, 2016-03-02:
Isn't fxload also available for arm architectures? Basically I only want to avoid depending on an external tool just because of 600 lines of c code that can also be integrated.

3esmit, 2016-03-02:
Yes, but the .sys hantek delivers are for x86 and 64 arch, none for arm. Or can I use x86 to extract .hex and use this hex in arm linux (raspbian, for example)?

davidgraeff, 2016-03-02:
Exactly. The firmware does not depend on the host system and is always the same. You may use the files that are located in firmware/.

closed 2017-12-17 by davidgraeff