OpenHantek/openhantek #23: OpenHantek master does not compile

hzsolt94, 2016-03-02:
I'm trying to compile OpenHantek on Ubuntu (both the master and the downloadable zip archive) and get the following errors during make :

/home/labor/Asztal/OpenHantek-openhantek-6026fd6/openhantek/src/main.cpp: In function ‘int main(int, char**)’:
/home/labor/Asztal/OpenHantek-openhantek-6026fd6/openhantek/src/main.cpp:45:70: error: ‘QMAKE_TRANSLATIONS_PATH’ was not declared in this scope
  openHantekTranslator.load("openhantek_" + QLocale::system().name(), QMAKE_TRANSLATIONS_PATH);
                                                                      ^
CMakeFiles/OpenHantek.dir/build.make:215: recipe for target 'CMakeFiles/OpenHantek.dir/src/main.o' failed
make[2]: *** [CMakeFiles/OpenHantek.dir/src/main.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/OpenHantek.dir/all' failed
make[1]: *** [CMakeFiles/OpenHantek.dir/all] Error 2
Makefile:116: recipe for target 'all' failed
make: *** [all] Error 2

Can you help me solving this?

hzsolt94, 2016-03-02:
I was able to compile and install this commit:
https://github.com/OpenHantek/openhantek/tree/3802257ab32f62e2491053428145f3ab578d312a
Altough it uses qmake, so it should be irrelevant.

davidgraeff, 2016-03-02:
I've changed the build system just a few hours ago. Continous integration also told me that it does not compile for linux. I'm on windows right now, but will fix it asap.

drothlis, 2016-03-03:
This should be fixed by ffeeee4.

closed 2016-03-03 by davidgraeff