OpenHantek/openhantek #25: More complete deb package

davidgraeff, 2016-03-03:
To make the debian package more "complete" and correct there are some file(INSTALL...) commands missing, all binaries should be installed to "bin", images to "share/openhantek" or similar etc. For debian and ubuntu the dependency names have to be figured out and added with one of the CPACK_ commands.

Theoretically we can also create a rpm, but again we have to work out the dependency names.

@drothlis CPack is used at the moment to create a debian package. Travis will issue the "package" target on every new tag and uploads all generated files in the package directory. I have no time for that right now, because i want to reintegrate parts of my other branch now. Would be cool if you look at this.

OliverHaag, 2016-03-03:
You could probably reuse some parts like dependencies from the old package: https://launchpad.net/~oliverhaag/+archive/ubuntu/openhantek
Sources are here: http://ppa.launchpad.net/oliverhaag/openhantek/ubuntu/pool/main/o/openhantek/

drothlis, 2016-03-07:
I spent about half a day creating a deb package but then I discovered sigrok which has a command-line tool (which is my actual goal -- capturing samples from the command line) so I'm unlikely to do any further work on this. For what it's worth the work so far is on my fork [here](https://github.com/drothlis/openhantek/commit/32c9c15). The remaining TODO list is:

> * Generate changelog (which is where debuild looks for the package's
>   version number) from a "changelog.in" file.
> * It isn't clear to me if the license is supposed to be GPL-3 or GPL-3+.
> * Fix CMakeLists.txt to install the binary into $prefix/bin, the images
>   into $prefix/share/openhantek/images, and not install readme, COPYING,
>   etc.
> * Rename firmware/90-hantek.rules to firmware/90-hantek.rules.in,
>   get CMakeLists.txt to generate firmware/90-hantek.rules with the
>   correct $prefix.
> * Get CMakeLists.txt to install the firmware & udev rule (into a
>   different destdir for the firmware package).
> * Script to build source package & push it to ppa?
>     * I wouldn't bother using cmake/cpack to generate the source
>       tarball, see e.g. https://cmake.org/Bug/view.php?id=6560 -- a simple `git archive` would do the trick.
> * Make travis build the deb packages?
> * Remove unnecessary dependencies from .travis.yml.

closed 2017-04-18 by davidgraeff