OpenHantek/openhantek #30: compile on debian jessie fails

juliusjunghans, 2016-04-25:
after some fiddeling around on another more default jessie system i got the software compiled (will document that later)

the firmware loader from:

compiled fine after adding some libraries

i "loaded" everything thats looks like firmware from the cd driver directory, but none seems to work.
or are those all warnings?

i bought the 6022BE online, maybe somebody can provide me with a older firmware?

./extractfw /home/julius/tmp/openhantek-code/Hantek6022BEAMD641.sys
BFD: /home/julius/tmp/openhantek-code/Hantek6022BEAMD641.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .text
BFD: /home/julius/tmp/openhantek-code/Hantek6022BEAMD641.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .rdata
BFD: /home/julius/tmp/openhantek-code/Hantek6022BEAMD641.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .data
BFD: /home/julius/tmp/openhantek-code/Hantek6022BEAMD641.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .pdata
Section .data found (starting at 0x1180, 10144 bytes)
Symbol _firmware found (offset 0xffffffffffffee80, 0 bytes)
Symbol _loader found (offset 0xffffffffffffee80, 10144 bytes)
Can't get firmware contents: Bad value

./extractfw /home/julius/tmp/openhantek-code/Hantek6022BEAMD642.sys
BFD: /home/julius/tmp/openhantek-code/Hantek6022BEAMD642.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .text
BFD: /home/julius/tmp/openhantek-code/Hantek6022BEAMD642.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .rdata
BFD: /home/julius/tmp/openhantek-code/Hantek6022BEAMD642.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .data
BFD: /home/julius/tmp/openhantek-code/Hantek6022BEAMD642.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .pdata
Section .data found (starting at 0x3500, 272 bytes)
Symbol _firmware found (offset 0xffffffffffffcb00, 0 bytes)
Symbol _loader found (offset 0xffffffffffffcb00, 272 bytes)
Can't get firmware contents: Bad value

ractfw# ./extractfw /home/julius/tmp/openhantek-code/Hantek6022BEIA641.sys
bfd_check_format: File format not recognized

./extractfw /home/julius/tmp/openhantek-code/Hantek6022BEIA642.sys
bfd_check_format: File format not recognized

./extractfw /home/julius/tmp/openhantek-code/Hantek6022BEX861.sys
BFD: /home/julius/tmp/openhantek-code/Hantek6022BEX861.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .text
BFD: /home/julius/tmp/openhantek-code/Hantek6022BEX861.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .rdata
BFD: /home/julius/tmp/openhantek-code/Hantek6022BEX861.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .data
Section .data found (starting at 0x0c00, 9864 bytes)
Symbol _firmware found (offset 0xfffffffffffff400, 0 bytes)
Symbol _loader found (offset 0xfffffffffffff400, 9864 bytes)
Can't get firmware contents: Bad value

./extractfw /home/julius/tmp/openhantek-code/Hantek6022BEX862.sys
BFD: /home/julius/tmp/openhantek-code/Hantek6022BEX862.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .text
BFD: /home/julius/tmp/openhantek-code/Hantek6022BEX862.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .rdata
BFD: /home/julius/tmp/openhantek-code/Hantek6022BEX862.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .data
Section .data found (starting at 0x2780, 8 bytes)
Symbol _firmware found (offset 0xffffffffffffd880, 0 bytes)
Symbol _loader found (offset 0xffffffffffffd880, 8 bytes)
Can't get firmware contents: Bad value

juliusjunghans, 2016-04-25:
how do i proceed?

davidgraeff, 2016-04-26:
OpenHantek is unfortunately not 6022BE compatible (no hardware trigger).

juliusjunghans, 2016-04-27:
oh, missed that.

whats the next cheapest usb oscillscope that openhantek can work with?
mine was about 70€.

or do you know other projects that work well with the 6022be?
there are a few entries on github but the first i tried did not work out of the box...
and why would it...as open source developer you can just release something have finished

davidgraeff, 2016-05-02:
Try www.sigrok.org :)

closed 2016-05-02 by davidgraeff