OpenHantek/openhantek #31: Merge firmware for Hantek 6022BE

mashu, 2016-05-31:
Hi,

Would it be possible to merge firmware from here and make Hantek 6022BE supported ?
https://github.com/rpcope1/Hantek6022API

Thanks,
Mateusz

jmatraszek, 2016-06-01:
Support for 6022 would be really cool!

mashu, 2016-06-01:
I managed to get it working on Linux, since Windows VM was crashing.
https://github.com/giomasce/openhantek
```
$ git branch
* (HEAD detached at origin/6022be)
  master
```
Some functionality is missing though, and not everything works. Qt5 crashes very often when I try to change some of options. Even  disabling channel crashes the app.
Besides that it's fairly usable.

hexdream, 2016-08-06:
Hi Mateusz, Jmatraszek,

Sorry for (potentially) ijacking this threadI am really struggling to get my 6022be to run under Ubuntu. I can manually load the firmware (I get the red light on the scope), but I cannot get OpenHantek to recognize the device. It keeps saying no hantek oscilloscope found. I have tried the normal branch of OpenHantek as well as the 6022 branch.  Can you suggest things I can look at? How does OpenHantek determine if a device is attached? Does the firmware version matter, and if it does, which one should I use? Is there a specific udev rule that I have to run?

Any suggestions would be welcome.

D

davidgraeff, 2016-08-09:
@hexdream You can always try to run OpenHantek as root, to exclude missing access rights / udev rules as a cause.
OpenHantek listens to the usb bus via the libusb library. On the first start it just enumerates all devices and looks for the right vendor/device id.

davidgraeff, 2017-04-18:
Will be fixed together with 6022 support

closed 2017-04-18 by davidgraeff