OpenHantek/openhantek #32: UI segfaults

igor-stoppa, 2016-06-17:
I'm getting a similar error message to what is described at the end of Issue #10 

This is the output:

    $ /usr/local/OpenHantek 
    Version 
    QObject::connect: No such slot OpenHantekMainWindow::sendCommand()
    QObject: Cannot create children for a parent that is in a different thread.
    (Parent is Hantek::Control(0x25568e0), parent's thread is QThread(0x247de30), current thread is Hantek::Control(0x25568e0)
    Segmentation fault (core dumped)

I'm using KDE on an up-to-date FC 23

I followed the instructions from the readme.md file against a freshly cloned master branch.

AndruPol, 2016-07-09:
Hi, I'm have same problem on ubuntu 14.04 lts

> andru@book:~/todo/hantek-dso2090/openhantek/build$ cmake ..
> The C compiler identification is GNU 4.8.4
>-- The CXX compiler identification is GNU 4.8.4
>-- Check for working C compiler: /usr/bin/cc
>-- Check for working C compiler: /usr/bin/cc -- works
>-- Detecting C compiler ABI info
>-- Detecting C compiler ABI info - done
>-- Detecting C compile features
>-- Detecting C compile features - done
>-- Check for working CXX compiler: /usr/bin/c++
>-- Check for working CXX compiler: /usr/bin/c++ -- works
>-- Detecting CXX compiler ABI info
>-- Detecting CXX compiler ABI info - done
>-- Detecting CXX compile features
>-- Detecting CXX compile features - done
>-- Found OpenGL: /usr/lib/i386-linux-gnu/libGL.so  
>-- Will not build translations! Please install Qt5LinguistTools
>-- Found libusb: /usr/lib/i386-linux-gnu/libusb-1.0.so  
>-- Looking for include file pthread.h
>-- Looking for include file pthread.h - found
>-- Looking for pthread_create
>-- Looking for pthread_create - not found
>-- Looking for pthread_create in pthreads
>-- Looking for pthread_create in pthreads - not found
>-- Looking for pthread_create in pthread
>-- Looking for pthread_create in pthread - found
>-- Found Threads: TRUE  
>-- Found FFTW: /usr/lib/i386-linux-gnu/libfftw3.so  
>-- Configuring done
>-- Generating done
>-- Build files have been written to: /home/andru/todo/hantek-dso2090/openhantek/build`

make

>andru@book:~/todo/hantek-dso2090/openhantek/build/openhantek$ ./OpenHantek
>Version 
>QObject::connect: No such slot OpenHantekMainWindow::sendCommand()
>QObject: Cannot create children for a parent that is in a different thread.
>(Parent is Hantek::Control(0x95a20d8), parent's thread is QThread(0x94777f8), current thread is 
Hantek::Control(0x95a20d8)
>09:13:32.943: Sending bulk command: 01 00 85 00 e8 ff ff d7 00 00 07 00
>09:13:32.943: Sending bulk command: 07 00 00 00 00 00 00 00
>09:13:32.944: Sending control command b4: 00 73 00 56 00 80 00 00 00 00 00 00 00 00 00 00 00
>09:13:32.991: Sending control command b5: 00 04 08 fd 20 40 ef 01 00 00 00 00 00 00 00 00 00`

can anybody help?

igor-stoppa, 2016-07-12:
I just finished building it for Ubuntu 16.04 and it seems to work.
The instructions, however, are incomplete: one needs to install also qt5base-dev and possibly qt5base-dev-tools (I installed both and the call to cmake started to suceed).

igor-stoppa, 2016-07-12:
The same HEAD works also with FC 24 (I have upgraded, since I filed this bug).

Also in this case, the instructions for rpm-based distro didn't fully work, as FC 24 doesn't have any libusb-1.0-0-devel

However, libusb-devel seem to be sufficient.

igor-stoppa, 2016-07-12:
If nobody disagrees, I'll close this issue in a couple of days.

laforest, 2016-07-22:
I see the exact same segfault here, when built from Git master branch, on a freshly upgraded Ubuntu 16.04.
Any required libsigrok components were also built from fresh Git repos.

braselectron, 2016-07-24:
I am also having segfault, no GUI (crashes) using Ubuntu 14.04 32bits (DSO-2250)

$ ./OpenHantek 
Version 
QObject::connect: No such slot OpenHantekMainWindow::sendCommand()
QObject: Cannot create children for a parent that is in a different thread.
(Parent is Hantek::Control(0x99e37c8), parent's thread is QThread(0x989a7f8), current thread is Hantek::Control(0x99e37c8)
22:54:28.903: Sending bulk command: 07 00 0a 00 00 00 00 00
22:54:28.904: Sending bulk command: 0b 00 02 00
22:54:28.905: Sending bulk command: 0c 00 02 00 00 00 00 00
22:54:28.907: Sending bulk command: 0d 00 01 00
22:54:28.908: Sending bulk command: 0e 00 02 00 f7 ff 00 00
22:54:28.909: Sending bulk command: 0f 00 ff ff 07 00 ff d7 07 00
22:54:28.911: Sending control command b4: 00 46 00 4d 00 7f 00 00 00 00 00 00 00 00 00 00 00
22:54:28.961: Sending control command b5: 00 04 f7 fd 20 bf ef 01 00 00 00 00 00 00 00 00 00
Segmentation fault (core dumped)

AndruPol, 2016-07-24:
I see OpenHantek GUI work on Ubuntu 14.04 32bits without DSO-2090 but crashes if DSO-2090 connected.
QObject: Cannot create children for a parent that is in a different thread.
(Parent is Hantek::Control(0x9acc648), parent's thread is QThread(0x995d7f8), current thread is Hantek::Control(0x9acc648)
07:30:49.774: Sending bulk command: 01 00 85 08 e8 ff ff d7 00 00 07 00
07:30:49.775: Sending bulk command: 07 00 00 00 00 00 00 00
07:30:49.776: Sending control command b4: 00 73 00 56 00 80 00 00 00 00 00 00 00 00 00 00 00
07:30:49.825: Sending control command b5: 00 04 08 02 20 40 ef 01 00 00 00 00 00 00 00 00 00
Ошибка сегментирования

igor-stoppa, 2016-10-16:
I just tried to rebuild on Ubuntu 16.10 and I also get the same segfault as reported above.

Version
QObject::connect: No such slot OpenHantekMainWindow::sendCommand()
QObject: Cannot create children for a parent that is in a different thread.
(Parent is Hantek::Control(0x561d5d72e690), parent's thread is QThread(0x561d5d645c10), current thread is Hantek::Control(0x561d5d72e690)
19:05:45.172: Sending bulk command: 01 00 85 02 e8 ff ff d7 00 00 07 00
19:05:45.174: Sending bulk command: 07 00 02 00 00 00 00 00
19:05:45.175: Sending control command b4: 00 4b 00 6e 00 ac 00 00 00 00 00 00 00 00 00 00 00
19:05:45.224: Sending control command b5: 00 04 08 02 20 40 10 01 00 00 00 00 00 00 00 00 00
Segmentation fault (core dumped)

I don't know if it makes any difference, but in my case the hex values shown are 64bit, while previous posts show 32 bit ones.

igor-stoppa, 2016-10-16:
I just tried the old .deb file from the ppa (it was built for maverick!) and it works.

I had missed the statement on the website: "The UI is written in C++/Qt 4 ..." an I was trying to build probably against Qt 5, so it's understandable that I would get these sort of errors.

This issue report is invalid.

In its place, we should have a feature request: Build against Qt 5.

closed 2016-10-16 by igor-stoppa