OpenHantek/openhantek #34: Ubuntu 14.04 LTS 32bits make errors

braselectron, 2016-07-22:
My System Software / Hardware Information

Hardware:
Processor: Intel Core 2 6300 @ 1.87GHz (2 Cores), 
Motherboard: Gigabyte 965P-DS3, 
Chipset: Intel 82P965/G965 + ICH8/R, 
Memory: 8192MB, 
Disk: 500GB Western Digital WD5000BPKT-0, 
Graphics: eVGA NVIDIA GeForce GT 610 2048MB, 
Audio: Realtek ALC888, 
Monitor: SyncMaster P2370, Native Resolution: 1920x1080
Network: Marvell 88E8056 PCI-E Gigabit

Software:
OS: Ubuntu 14.04, 
Kernel: 3.13.0-91-generic (i686), 
Desktop: Unity 7.2.6, 
Display Server: X Server 1.15.1, 
Display Driver: nouveau 1.0.10, 
OpenGL: 3.3 
Mesa 10.1.3 Gallium 0.4, 
Compiler: GCC 4.8 + Clang 3.4-1ubuntu3, 
File-System: ext4, Screen 

Using:
binutils-dev     2.24-5ubuntu i386         GNU binary utilities (BFD developm
cmake             2.8.12.2-0ub i386         cross-platform, open-source make s
libfftw3-dev:i    3.3.3-7ubunt i386         Library for computing Fast Fourier
libusb-1.0-0-d  2:1.0.17-1ub i386         userspace USB programming library 
qttools5-dev-t  5.2.1-8build i386           Qt 5 development tools
autoconf          2.69-6       all                automatic configure script builder
automake        1:1.14.1-2ub all            Tool for generating GNU Standards-
build-essentia  11.6ubuntu6  i386        Informational list of build-essent
subversion       1.8.8-1ubunt i386         Advanced version control system
unrar                1:5.0.10-1ub i386         Unarchiver for .rar files (non-fre

The git clone was downloaded and extracted in the home: ~/openhantek/

I had to edit CMakeLists.txt and remove:
`cmake_minimum_required(VERSION 3.0 FATAL_ERROR)`
and use:
`cmake_minimum_required(VERSION 2.8.12 FATAL_ERROR)`

and also I changed this (but I believe does not matter much):

  #  C++11: Workaround for cmake < 3.1 with no target_compile_features support.
  if (CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")

    # Patch for Ubuntu 14.04 LTS with cmake 2.8.12
    ################################################
    # Find includes in corresponding build directories
    set(CMAKE_INCLUDE_CURRENT_DIR ON)
    # Instruct CMake to run moc automatically when needed.
    set(CMAKE_AUTOMOC ON)

    # Find the QtWidgets library
    find_package(Qt5Widgets)

    set(CMAKE_CXX_STANDARD 14)
    #end patch
    ################################################
  endif()

than `cmake ../`  worked but `make` failed.

:~/openhantek/build$ make
[  4%] Automoc for target OpenHantek
[  4%] Built target OpenHantek_automoc
Linking CXX executable OpenHantek
CMakeFiles/OpenHantek.dir/src/main.cpp.o: In function "main":
~/openhantek/openhantek/src/main.cpp:37: undefined reference to "qInitResources_application()"
collect2: error: ld returned 1 exit status
make[2]: *** [openhantek/OpenHantek] Error 1
make[1]: *** [openhantek/CMakeFiles/OpenHantek.dir/all] Error 2
make: *** [all] Error 2

When I removed the line with: `Q_INIT_RESOURCE(application);`  from main.cpp the `make` completed and the executable was created,

make
[  4%] Automoc for target OpenHantek
[  4%] Built target OpenHantek_automoc
[  4%] Built target OpenHantek

(only 4% ?!?!? but executable is created)

 but when I run it the device clicks once and I get the following output (gui crashes):

~/openhantek/build/openhantek$ ./OpenHantek

Version 
QObject::connect: No such slot OpenHantekMainWindow::sendCommand()
QObject: Cannot create children for a parent that is in a different thread.
(Parent is Hantek::Control(0x963d928), parent's thread is QThread(0x94f47c8), current thread is Hantek::Control(0x963d928)
01:36:55.138: Sending bulk command: 07 00 00 00 00 00 00 00
01:36:55.139: Sending bulk command: 0b 00 00 00
01:36:55.140: Sending bulk command: 0c 00 02 00 00 00 00 00
01:36:55.141: Sending bulk command: 0d 00 01 00
01:36:55.142: Sending bulk command: 0e 00 02 00 30 fc 00 00
01:36:55.143: Sending bulk command: 0f 00 ff ff 07 00 ff d7 07 00
01:36:55.143: Sending control command b4: 00 4f 00 5f 00 7f 00 00 00 00 00 00 00 00 00 00 00
01:36:55.191: Sending control command b5: 00 04 08 fd 20 40 ef 01 00 00 00 00 00 00 00 00 00
Segmentation fault (core dumped)

I need help.

braselectron, 2016-07-24:
So I started again.  

1. Removed my local local clone of openhantek (rmdir ~/openhantek)
2. downloaded the binary of cmake 3.5.2  to /opt
    curl -sSL https://cmake.org/files/v3.5/cmake-3.5.2-Linux-i386.tar.gz | sudo tar -xzC /opt
3. cd into ~/openhantek ;  mkdir build ; cd into build
4. /opt/cmake-3.5.2-Linux-i386/bin/cmake ../   # no errors this time
5. cd into ~/openhantek/build/openhantek to execute the binary ./OpenHantek

Getting the following output:

Version 
QObject::connect: No such slot OpenHantekMainWindow::sendCommand()
QObject: Cannot create children for a parent that is in a different thread.
(Parent is Hantek::Control(0x99e37c8), parent's thread is QThread(0x989a7f8), current thread is Hantek::Control(0x99e37c8)
22:54:28.903: Sending bulk command: 07 00 0a 00 00 00 00 00
22:54:28.904: Sending bulk command: 0b 00 02 00
22:54:28.905: Sending bulk command: 0c 00 02 00 00 00 00 00
22:54:28.907: Sending bulk command: 0d 00 01 00
22:54:28.908: Sending bulk command: 0e 00 02 00 f7 ff 00 00
22:54:28.909: Sending bulk command: 0f 00 ff ff 07 00 ff d7 07 00
22:54:28.911: Sending control command b4: 00 46 00 4d 00 7f 00 00 00 00 00 00 00 00 00 00 00
22:54:28.961: Sending control command b5: 00 04 f7 fd 20 bf ef 01 00 00 00 00 00 00 00 00 00
Segmentation fault (core dumped)

I when to https://sourceforge.net/projects/openhantek/ (older version)
Complied from source using qmake (qt5) and got screen of the GUI, my DSO-2250 clicks when changing voltage levels but to output.

I would appreciate any help from openhantek community.

braselectron, 2016-07-24:
Now the **strange** "facts".

I found this page: 
[http://verahill.blogspot.com.br/2012/12/298-hantek-dso-2250-usb-with-openhantek.html](http://verahill.blogspot.com.br/2012/12/298-hantek-dso-2250-usb-with-openhantek.html)

The instructions told me to first install dependence packages with:

`sudo apt-get install libqt4-dev libfftw3-dev qt4-qmake fxload libusb-1.0-0-dev libbfd-dev`

but I had all already installed (just to be sure, check using `dpkg -l <package list>`)

Then I downloaded the project (older version - I believe 'patched' for DSO-2250) with:

`svn checkout http://svn.code.sf.net/p/openhantek/code/trunk openhantek-code`

then go to the directory created:

`cd openhantek-code`

Take additional care by using the correct 'qmake' version, which is qt4 (not qt5)

You can check your versions installed to find the qmake(qt4) binary, using: 

`locate -b '\qmake' | xargs -ri find {} -prune -type f -executable`

Since I have two versions installed I need to issue:

`/usr/share/qt4/bin/qmake`  (I had to use the 'locate' result for the path)

At the end of qmake 'no errors', just a slient ending (just the prompt returned)

Then I proceded to the 'make' with:

`make`

which also ended with no errors messages. Great! So there must be a binary. 

`cd bin`

execute the binary with (because I did not 'install' it yet):
`./openhantek`

And the GUI pop-up on screen! Hurray! It worked!!!

Please note that I had already installed the firmware part (please check instructions).

And, by the way, the only firmware extraction source that I was able to build was using:

[https://sourceforge.net/projects/openhantek/files/openhantek-extractfw/openhantek-extractfw-0.7.0.tar.bz2/download](https://sourceforge.net/projects/openhantek/files/openhantek-extractfw/openhantek-extractfw-0.7.0.tar.bz2/download)

After downloading it go to the directory and build it with:

1)  `./configure`
2)  `make`

Than put the DSOxxxx1.sys (windows driver file - check instructions) in the same directory and issues the command:

`./openhantek-extractfw DSOxxxx1.sys`

Be aware that the name of the resulting files must be the same as used in the file: "**90-hantek.rules**"

Also now it worked again!!!  Two files produced.  **Remember, follow the instructions....**

**So my question to OpenHantek community is:** why is it that the project participants here are making it so difficult for the unskilled users to **'make' this a simple and usable piece of software** ?

I believe that the original goal of GNU and FSF is to make software accessible to everyone, is to get the job done, is to empower the people, and to make things simple. Cheers!

Now how can I get this newer version, available here, to work?

OliverHaag, 2016-07-24:
I'm not really working on OpenHantek myself anymore and didn't try the github master for some time. @davidgraeff should have a better overview here but I would really prefer to keep the master branch clean and in a buildable and useable state at any time (like the old one on sf.net svn was) for the reasons @braselectron mentioned.

braselectron, 2016-07-25:
Dear Oliver Haag,

First, my compliments: you are a "master" coder!  
Please excuse my english.  Thank for your work and for the promptly answer.

Just to add more (debug) information here, I just noticed now with my "tests" with the sf.net.svn version that worked that the voltage level looks good (needs calibration), but the FFT function is unstable (crashes).  I can get only a quick image of the FFT output that immediately vanishes.  I have to click on the FFT CHx bottons many times in combination with others to make it show the quick image overlay.  

[image]
[image]
[image]

My console outputs this: 
~/openhantek-code/openhantek/bin$ ./openhantek
QObject: Cannot create children for a parent that is in a different thread.
(Parent is Hantek::Control(0x950f3d8), parent's thread is QThread(0x939b0b8), current thread is Hantek::Control(0x950f3d8)

Also the "record length" in the "Roll" option is unstable.  If you change it and return to "Roll" position again the signal image is incorrect.  You need to restart the binary to recover.

So I conclude that I have still issues with even the sf.net.svn (i.e. 0.2.0x version), or I my be having problems in the build process for my installation layout (or dependencies).

I really would like to avoid the VirtualBox solution.  So I need help.

Following what @OliverHaag said, if I can make a suggestion, we could have a directory with working versions/releases  (tar clones and also binaries for the main distributions - i386 and 64bits) which I believe would better help the community that only needs a working version for faster results.

Also a good wiki to help newbies with step-by-step instructions (not to say also as a reference to developers to keep build and dependency history for each release).  Sorry, but we "engineers" like to bring order, to chaos theory. 

A good add-on would be to have a "plugin" subsystem, which the community would help build by making new functions for the openhantek project, I believe would be a greater success, and permit a more stable core.