OpenHantek/openhantek #38: Add support for Qt 5

igor-stoppa, 2016-10-16:
Most distros ship with Qt 5 and that's the default build environment used, when trying to build from sources.

It would be nice to have support for it.

Should that not be possible, I think it would be good to at least detect that the Qt version is different from the expected one (5 instead of 4), stop the build and generate a meaningful error message.

davidgraeff, 2016-10-19:
Hm, actually I build the sources with Qt5 successfully as far as I can remember and travis builds with Qt5, too. Are u sure it is not working for you with Qt5?

igor-stoppa, 2016-10-19:
Apologies, I was not sufficiently clear. It does build, but when I try to run it with the scope attached, it fails, in the way reported in https://github.com/OpenHantek/openhantek/issues/32.

closed 2017-04-18 by davidgraeff