OpenHantek/openhantek #41: Compilation Errors on Mint 18

krakenrad, 2016-11-09:
Hi,

I needed to install 

Can you help me compile openhantek? I'm getting this error and am new to cmake:

$ cmake ../
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Found libusb: /usr/lib/x86_64-linux-gnu/libusb-1.0.so  
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found FFTW: /usr/lib/x86_64-linux-gnu/libfftw3.so  
CMake Error at cmake/CPackInfos.cmake:28 (string):
  string sub-command REPLACE requires at least four arguments.
Call Stack (most recent call first):
  CMakeLists.txt:30 (include)

CMake Error at cmake/CPackInfos.cmake:29 (string):
  string sub-command REPLACE requires at least four arguments.
Call Stack (most recent call first):
  CMakeLists.txt:30 (include)

CMake Error at cmake/CPackInfos.cmake:30 (string):
  string sub-command REPLACE requires at least four arguments.
Call Stack (most recent call first):
  CMakeLists.txt:30 (include)

CMake Error at cmake/CPackInfos.cmake:31 (string):
  string sub-command REPLACE requires at least four arguments.
Call Stack (most recent call first):
  CMakeLists.txt:30 (include)

CMake Error at cmake/CPackInfos.cmake:32 (list):
  list GET given empty list
Call Stack (most recent call first):
  CMakeLists.txt:30 (include)

-- Configuring incomplete, errors occurred!
See also "/home/jam/HantekBuild/openhantek-master/build/CMakeFiles/CMakeOutput.log".
See also "/home/jam/HantekBuild/openhantek-master/build/CMakeFiles/CMakeError.log".

saogalde, 2016-11-20:
This really looks like this issue: https://github.com/OpenHantek/openhantek/issues/39 , although you don't have the two lines with the git error. Anyway, it can be useful.

davidgraeff, 2017-04-18:
fixed now

closed 2017-04-18 by davidgraeff

rahul-c54, 2019-07-15:
> fixed now

can you tell me how to fix it?