OpenHantek/openhantek #43: Big white square in GUI when deactivating toolbars

saogalde, 2016-11-20:
Hi everyone,
I came up with the following bug, which is happening in Qt5.7.0 in OSX Yosemite. When I deactivate all toolbars in OpenHantek and re-open the program, a big white square appears. This doesn't happen  at the time when I deactivate all toolbars, but when I reopen the program with these settings. It can be easily solved by reactivating all toolbars again.

Normal screenshot
[image]

Screenshot when error happening
[image]

# Console log
```
saogalde@macpro: ~/Downloads/openhantek-master/build/openhantek$ ./OpenHantek
Version 
QObject::connect: No such slot OpenHantekMainWindow::sendCommand()
QWidgetWindow(0x7ffe02f30ed0, name="QToolBarClassWindow") ( QScreen(0x7ffe02d1e1c0, name="Color LCD") ): Attempt to set a screen on a child window.
QWidgetWindow(0x7ffe02f36da0, name="QToolBarClassWindow") ( QScreen(0x7ffe02d1e1c0, name="Color LCD") ): Attempt to set a screen on a child window.
QWidgetWindow(0x7ffe02f36e00, name="QToolBarClassWindow") ( QScreen(0x7ffe02d1e1c0, name="Color LCD") ): Attempt to set a screen on a child window.
QWidgetWindow(0x7ffe02f30ed0, name="QToolBarClassWindow") ( QScreen(0x7ffe02d1e1c0, name="Color LCD") ): Attempt to set a screen on a child window.
QWidgetWindow(0x7ffe02f36da0, name="QToolBarClassWindow") ( QScreen(0x7ffe02d1e1c0, name="Color LCD") ): Attempt to set a screen on a child window.
QWidgetWindow(0x7ffe02f36e00, name="QToolBarClassWindow") ( QScreen(0x7ffe02d1e1c0, name="Color LCD") ): Attempt to set a screen on a child window.
QWidgetWindow(0x7ffe02f36e00, name="QToolBarClassWindow") ( QScreen(0x7ffe02d1e1c0, name="Color LCD") ): Attempt to set a screen on a child window.
QWidgetWindow(0x7ffe02f36da0, name="QToolBarClassWindow") ( QScreen(0x7ffe02d1e1c0, name="Color LCD") ): Attempt to set a screen on a child window.
QWidgetWindow(0x7ffe02f30ed0, name="QToolBarClassWindow") ( QScreen(0x7ffe02d1e1c0, name="Color LCD") ): Attempt to set a screen on a child window.
QWidgetWindow(0x7ffe02f36da0, name="QToolBarClassWindow") ( QScreen(0x7ffe02d1e1c0, name="Color LCD") ): Attempt to set a screen on a child window.
```

davidgraeff, 2017-04-18:
Related to #57 I guess

davidgraeff, 2017-12-17:
Should be fixed now if Qt >= 5.4 is used for compiling

closed 2017-12-17 by davidgraeff