OpenHantek/openhantek #46: Hantek 1008c - 8 channel oscilloscope support

szg0000, 2017-01-04:
Does openhantek support "Hantek 1008c" 8 channel oscilloscope?

mmahkamov, 2017-01-13:
I just bought Hantek 1008c. Will check if it's supported.

rozap, 2017-02-25:
@mmahkamov any progress with the 1008c?

szg0000, 2018-09-23:
Any update with Hantek 1008c?

davidgraeff, 2018-09-24:
Support for the Hantek 1008c will not magically appear :) I do not own such a device.

laci200270, 2018-10-31:
@davidgraeff  may we help you with usb pcap dumps? or what information do you need?

christianlupus, 2019-02-17:
> Support for the Hantek 1008c will not magically appear :) I do not own such a device.

I have a Hantek 1008C ready here. I could sent it to you, if this helps (I am also from Germany). PLease tell me, if I should. Then we can discuss about shipping separately.

davidgraeff, 2019-02-17:
See https://github.com/OpenHantek/openhantek/issues/277 :)

Gurom, 2019-10-13:
it could be good to have openhuntek for 1008C

christianlupus, 2019-10-16:
> it could be good to have openhuntek for 1008C

Just read #277 and you will get your answer.