OpenHantek/openhantek #47: fftw issues during installation

mmaboharra, 2017-01-09:
hi I've got the following errors during installation.
what version of fftw should I use?

usr@pc:~/Downloads/OpenHantek-openhantek-6a7a942/build$ cmake ../
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Found libusb: /usr/lib/x86_64-linux-gnu/libusb-1.0.so  
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found FFTW: /usr/lib/x86_64-linux-gnu/libfftw3.so  
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
CMake Error at cmake/CPackInfos.cmake:28 (string):
  string sub-command REPLACE requires at least four arguments.
Call Stack (most recent call first):
  CMakeLists.txt:30 (include)

CMake Error at cmake/CPackInfos.cmake:29 (string):
  string sub-command REPLACE requires at least four arguments.
Call Stack (most recent call first):
  CMakeLists.txt:30 (include)

CMake Error at cmake/CPackInfos.cmake:30 (string):
  string sub-command REPLACE requires at least four arguments.
Call Stack (most recent call first):
  CMakeLists.txt:30 (include)

CMake Error at cmake/CPackInfos.cmake:31 (string):
  string sub-command REPLACE requires at least four arguments.
Call Stack (most recent call first):
  CMakeLists.txt:30 (include)

CMake Error at cmake/CPackInfos.cmake:32 (list):
  list GET given empty list
Call Stack (most recent call first):
  CMakeLists.txt:30 (include)

-- Configuring incomplete, errors occurred!

yboumenir, 2017-02-19:
Hey I had the same error, but thats because you didnt get the one from git (i.e. i think its bc you downloaded tar one from his website --  and didnt git clone this one).

davidgraeff, 2017-04-18:
This problem is fixed now, thanks for reporting.

closed 2017-04-18 by davidgraeff