OpenHantek/openhantek #48: INSTRUSTAR and OpenHantek - possible?

denmodjo, 2017-02-01:
Hello.
I have a USB oscilloscope INSTRUSTAR-205A (http://english.instrustar.com/). 
Is it possible to work OpenHantek with him in theory?
What will it take to finalize? The name of the USB chip, ADC? The firmware supplied drivers (how to merge?)?
[Здравствуйте.
У меня USB осцилограф INSTRUSTAR-205A (http://english.instrustar.com/). 
Возможна ли работа OpenHantek с ним в теории?
Что потребуется для доработки? Название чипа USB, ADC? Прошивка из поставляемых драйверов (как ее слить?)?]

closed 2017-12-17 by davidgraeff

andykarpov, 2017-12-20:
Hm, it's seems ISDS205A has almost the same hardware like Hantek 6022BE.
At least the usb chip CY7C68013A and ADC AD9288.
The only difference is inside the EEPROM / attenuators connection, probably.
Why "wontfix" has been applied to this task/request?

davidgraeff, 2017-12-20:
Ok, I didn't know that. But we also can't fill the bug tracker with device requests. If someone is willing to add the device via PR, then that's another case.
* The firmware goes to openhantek/res/firmware
* The openhantek/src/hantek/models.h needs the model information
* The openhantek/src/hantek/hantekdsocontrol.cpp needs the implementation

That's it, if the devices firmware speaks the hantek protocol at least.

andykarpov, 2017-12-20:
Ok, **davidgraeff**, thanks a lot for the explanations!
The only way to add support to ISDS205A is to write it yourself then :)
Ok, challenge accepted! :) It could take a loooot of time, but i will try, then we'll see ;)
I'll try a github way: fork -> blablablah -> pull request.