OpenHantek/openhantek #49: DSO-5200 calibrator signal not right

alaindef, 2017-03-02:
[image].
testing openhantek ubuntu on DSO-5200.
the display seems to skip pulses when I test the calibrator signal available at the back (see attachment). I guess this is a firmware issue.  I use the dso5200x86-firmware.hex and dso5200x86-loader.hex files I found on this site. 
I want to try extracting the hex files from my working windows 10- 64 bit application, but I can't get the executable for extractfw. I try to compile the extractfw.c file, but got errors "undefined reference to bfd_xxx". I am not experienced in gcc though.
When this firmware issue is solved I 'd like to work on the sw. Can somebody help?

alain.de.feyter@gmail.com

davidgraeff, 2018-01-06:
Firmware uploading is now part of Openhantek. Would you please test your measurement again?

alaindef, 2018-01-06:
Thanks for your comment, David. It has been a while since I worked on this subject, but I plan to do new tests mid February.
Alain

davidgraeff, 2018-03-09:
Closing for now. Please also test the development branch

closed 2018-03-09 by davidgraeff