OpenHantek/openhantek #50: building from source, Ubuntu 16.04

stawel, 2017-03-10:
hi, 
I was struggling with the build process on Ubuntu 16.04,
I got this error:
```
CMake Error at openhantek/CMakeLists.txt:3 (find_package):
  By not providing "FindQt5Widgets.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "Qt5Widgets", but CMake did not find one.

  Could not find a package configuration file provided by "Qt5Widgets" with
  any of the following names:

    Qt5WidgetsConfig.cmake
    qt5widgets-config.cmake

  Add the installation prefix of "Qt5Widgets" to CMAKE_PREFIX_PATH or set
  "Qt5Widgets_DIR" to a directory containing one of the above files.  If
  "Qt5Widgets" provides a separate development package or SDK, be sure it has
  been installed.
```

It looks like I need to install also:
sudo apt-get install qttools5-dev

And the install location looks quite weird:
```
sudo make install
[  4%] Automatic moc, uic and rcc for target OpenHantek
[  4%] Built target OpenHantek_automoc
[100%] Built target OpenHantek
Install the project...
-- Install configuration: "RelWithDebInfo"
-- Installing: /usr/local/./ChangeLog
-- Installing: /usr/local/./COPYING
-- Installing: /usr/local/./readme.md
-- Installing: /usr/local/./OpenHantek
-- Installing: /usr/local/images/openhantek.png
-- Installing: /usr/local/images/analysis.png
-- Installing: /usr/local/images/scope.png
-- Installing: /usr/local/images/options.png
-- Installing: /usr/local/images/colors.png
-- Installing: /usr/local/images/files.png
-- Installing: /usr/local/images/stop.png
-- Installing: /usr/local/images/start.png
-- Installing: /usr/local/images/print.png
-- Installing: /usr/local/images/save.png
-- Installing: /usr/local/images/digitalphosphor.png
-- Installing: /usr/local/images/open.png
-- Installing: /usr/local/images/zoom.png
-- Installing: /usr/local/images/save-as.png
-- Installing: /usr/local/images/export-as.png
-- Installing: /usr/local/images/cursors.png
-- Installing: /usr/local/images/openhantek.icns
-- Installing: /usr/local/images/analysis.svg
-- Installing: /usr/local/images/options.svg
-- Installing: /usr/local/images/digitalphosphor.svg
-- Installing: /usr/local/images/openhantek.svg

```

Best regards,
Paweł

efiLabs, 2017-03-29:
hi Pawel :

i'm not from the openhantek team, but maybe i can help

sorry, i didn't really try to follow your issue, but i was able to compile the source without problems ... have a look at my link which contains the project and keep in mind that i made already changes to some files and added my Console debug monitor (maybe u don't want it) as well as my utypes.h which is needed by Console.h/cpp

fetch the latest QT 5.8 (i use 5.71) from the qt website and this should be it

https://github.com/OpenHantek/openhantek/issues/55

cheers Klaus

davidgraeff, 2017-04-18:
fixed

closed 2017-04-18 by davidgraeff

skander36, 2017-08-06:
Hi I try to install DSO 6022BE but I getting this error in cmake gui -

The CXX compiler identification is GNU 6.3.0
The C compiler identification is GNU 6.3.0
Check for working CXX compiler: /usr/bin/c++
Check for working CXX compiler: /usr/bin/c++ -- works
Detecting CXX compiler ABI info
Detecting CXX compiler ABI info - done
Detecting CXX compile features
Detecting CXX compile features - done
Check for working C compiler: /usr/bin/cc
Check for working C compiler: /usr/bin/cc -- works
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Detecting C compile features
Detecting C compile features - done
Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
CMake Warning (dev) at CMakeLists.txt:23 (add_definitions):
  Policy CMP0005 is not set: Preprocessor definition values are now escaped
  automatically.  Run "cmake --help-policy CMP0005" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at CMakeLists.txt:31 (find_package):
  By not providing "Findlibusb.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "libusb", but
  CMake did not find one.

  Could not find a package configuration file provided by "libusb" with any
  of the following names:

    libusbConfig.cmake
    libusb-config.cmake

  Add the installation prefix of "libusb" to CMAKE_PREFIX_PATH or set
  "libusb_DIR" to a directory containing one of the above files.  If "libusb"
  provides a separate development package or SDK, be sure it has been
  installed.

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.7)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

Configuring incomplete, errors occurred!
See also "/home/robert/openhantek/build/CMakeFiles/CMakeOutput.log".

davidgraeff, 2017-08-06:
You need libusb1.0 to be installed especially the source packet as well

skander36, 2017-08-09:
Ok ,
Thank you for your kindly response .

By,

Robert