OpenHantek/openhantek #57: Blank display on OSX

DanielO, 2017-04-04:
Hi,
I'm trying to build OpenHantek on OSX (w/ MacPort) and it does work, although I had to modify ./openhantek/CMakeFiles/OpenHantek.dir/flags.make and add -I/opt/local/include to the end of CXX_INCLUDES.

However when I run it I don't get any traces, the output is as follows..
```[Maelstrom 10:05] ~/projects/openhantek/build >./openhantek/OpenHantek
Version
QObject::connect: No such slot OpenHantekMainWindow::sendCommand()
Warning: This Hantek DSO model isn't supported officially, so it may not be working as expected. Reports about your experiences are very welcome though (Please open a feature request in the tracker at https://sf.net/projects/openhantek/ or email me directly to oliver.haag@gmail.com). If it's working perfectly I can remove this warning, if not it should be possible to get it working with your help soon.
QObject: Cannot create children for a parent that is in a different thread.
(Parent is Hantek::Control(0x7f8a5d5938e0), parent's thread is QThread(0x7f8a5d401620), current thread is Hantek::Control(0x7f8a5d5938e0)
QWidgetWindow(0x7f8a6232dcd0, name="QToolBarClassWindow") ( QScreen(0x7f8a5d7a2230, name="Color LCD") ): Attempt to set a screen on a child window.
QWidgetWindow(0x7f8a5fe89580, name="QToolBarClassWindow") ( QScreen(0x7f8a5d7a2230, name="Color LCD") ): Attempt to set a screen on a child window.
QWidgetWindow(0x7f8a5fbed4c0, name="QToolBarClassWindow") ( QScreen(0x7f8a5d7a2230, name="Color LCD") ): Attempt to set a screen on a child window.
10:05:18.324: Sending bulk command: 01 00 85 02 01 00 ff d7 00 00 07 00
10:05:18.325: Sending bulk command: 07 00 02 00 00 00 00 00
10:05:18.325: Sending control command b4: 00 47 00 63 00 7f 00 00 00 00 00 00 00 00 00 00 00
10:05:18.373: Sending control command b5: 00 04 08 fd 20 40 ef 01 00 00 00 00 00 00 00 00 00
10:05:18.460: Capture state changed to 2
10:05:18.467: Received 20480 B of sampling data
10:05:18.468: Starting to capture
10:05:19.423: Capture state changed to 0
10:05:19.424: Enabling trigger
10:05:24.224: Starting to capture
10:05:25.228: Enabling trigger
10:05:30.227: Starting to capture
10:05:31.224: Enabling trigger
10:05:39.228: Forcing trigger
10:05:40.226: Capture state changed to 1
```

The window itself looks like..
[image]

This is using Qt 5.6.2, i.e.
```[Maelstrom 10:07] ~/projects/openhantek/build >otool -L ./openhantek/OpenHantek
./openhantek/OpenHantek:
	/opt/local/libexec/qt5/lib/QtPrintSupport.framework/Versions/5/QtPrintSupport (compatibility version 5.6.0, current version 5.6.2)
	/opt/local/libexec/qt5/lib/QtOpenGL.framework/Versions/5/QtOpenGL (compatibility version 5.6.0, current version 5.6.2)
	/System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL (compatibility version 1.0.0, current version 1.0.0)
	/opt/local/lib/libusb-1.0.0.dylib (compatibility version 2.0.0, current version 2.0.0)
	/opt/local/lib/libfftw3.3.dylib (compatibility version 9.0.0, current version 9.5.0)
	/opt/local/libexec/qt5/lib/QtWidgets.framework/Versions/5/QtWidgets (compatibility version 5.6.0, current version 5.6.2)
	/opt/local/libexec/qt5/lib/QtGui.framework/Versions/5/QtGui (compatibility version 5.6.0, current version 5.6.2)
	/opt/local/libexec/qt5/lib/QtCore.framework/Versions/5/QtCore (compatibility version 5.6.0, current version 5.6.2)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 307.5.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1238.50.2)
```

davidgraeff, 2017-04-18:
The content part is drawn via Qt/OpenGL while the rest of the window is drawn with traditional Qt methods. Please try to start the application with the environment variable QT_GRAPHICSSYSTEM set to 'raster'. Other options are 'native' and 'opengl'.

`QT_GRAPHICSSYSTEM=raster OpenHantek`

DanielO, 2017-04-18:
I tried all 3 but observed no change in behaviour.

ta33ers, 2017-07-29:
I would also like to add that I am also having trouble:
Version 
QObject: Cannot create children for a parent that is in a different thread.
(Parent is Hantek::Control(0xa0e5e0), parent's thread is QThread(0x933c10), current thread is Hantek::Control(0xa0e5e0)
This is both on Arch, and Mint Linux

tildebyte, 2017-08-12:
If you tried to copypasta that, it won't change anything. You need to do
`QT_GRAPHICSSYSTEM=raster; OpenHantek`
(note the semicolon)

DanielO, 2017-08-12:
I ran it like so

env QT_GRAPHICSSYSTEM=raster ./openhantek/OpenHantek

With no change in behaviour.

davidgraeff, 2017-12-17:
Should be fixed now if Qt >= 5.4 is used for compiling.

closed 2017-12-17 by davidgraeff