OpenHantek/openhantek #62: No probe settings?

daybyter, 2017-04-23:
There is no setting for the probes (x1, x10 etc) yet?

Would be just a dropdown menu and a multiplication of the grid units?

KimBP, 2017-04-23:
I'm actually looking into it right now.
Unfortunately I think it is a bit more complicated than what you suggest. 
If you read section 7 (Some 6022 Peculiarities) of the manual available here: http://www.eevblog.com/forum/testgear/hantek-6022be-20mhz-usb-dso/msg1000844/#msg1000844 you'll see it is a very limited voltage range the scope can be used (if you want to avoid clipping or want to avoid zoom)
Clipping is the worst of the two because you don't really know if it how the signal actually looks like or you have just reached the limit of the scopes ADC.

For now I've added (locally) a x10 checkbox to the two channels. I'm planning to have the gain drop down follow this checkbox. I.e. if not checked only available gains are 100mV, 200mV, 500mv and 1V and if checked it will be 1V, 2V, 5V and 10V respectively.

I still haven't looked into whether the current implementation actually do command the scope correctly - but hopefully that's in place so it is only the dropdown I have to change.

Will let you know when I have something to play with.

daybyter, 2017-04-24:
I know about the clipping issue.
Would be cool to have some other probes and settings for the 6022 bzw. With a x3 probe, one could measure a 12v signal with a good resolution.
I think the rigol 1054z has a nice number of probe settings as an example. 1,2,5,10,... steps.

KimBP, 2017-04-24:
In that case I ought maybe go for a dropdown and not a checkbox

daybyter, 2017-04-24:
I'd say x1,x10,x100,x1000 are common values. So a dropdown makes more sense, I guess.

If you need coding help, let me know. Have some C++ experience, but not much QT experience yet.

dendvz, 2017-05-11:
Personally I would vote for the x1 / x10 checkbox. Maybe accompanied by probe config dialog somewhere in the settings menu.

Reason behind that is quite simple, the [ x1 / x10 ] **attenuator** is selected by slider on the probe itself.  So it is quite natural to move slider and select / unselect checkbox with a single mouse click. All other attenuators (100:1, 1000:1, 20:1/200:1) are only available if purchased separately.

@daybyter, those 1, 2, 5, 10... steps are specifically designed for e.g. Testec TT-SI 9002 which has slider selection 20:1 / 200:1  http://www.testec.de/en/products/product-categories/differrential-probes/ . It can be simulated by x1 / x10 front panel checkbox + x20 '**external attenuator**' hidden in the menus.

Thanks.
Denis.

jedi7, 2017-08-25:
I vote for dropdown. I have (for example) probes like 1x, 10x, 100x.
I wanted to implement it myself, but if you already working on it, then I will wait :)

dendvz, 2017-08-25:
@jedi7, do these probes have multi-position sliders? If not, I would still prefer probe config dialog (plus maybe a checkbox / radiobuttons on front panel to reflect the position of a slider).

You can find more explanations in my comment to #70; there are several major use-cases, and "classical oscilloscope" one presumes minimalistic front panel style. The drop-down responds to mouse scroll events, so you can accidentally change the settings.

jedi7, 2017-08-25:
@dendvz And what about different style? Like I described in #74 ?

Btw I suppose, the oscilloscope uses technicians which need detailed settings, then I prefer to have more settings than minimalist. :)

And I have 3 probes, a) 1x , b) 1x-10x, c) 100x

dendvz, 2017-08-25:
@jedi7 Your example of recalculating voltage into current #74 does not necessarily require math channel, it could be just another kind of probe which output signal has display units of e.g. mA/div. 

An actual difference between "oscilloscope for technicians" and "data acquisition system for scientists" is the same as difference between fixed-point and floating-point number representation. As a technician, I am expecting predictable setup: "the waveform on pin X should fit into 4 x 4 grid cells, if timebase is 100us/div and voltage is 500mV/div". As a scientist, I am expecting almost an opposite: "show me the signal at best available resolution and draw the reasonable gridlines". As a summary, I am not trying to minimize the total number of settings, only want to keep them better organized. Have I already said "skins"?

jedi7, 2017-08-26:
@dendvz ok ok :) skins sounds great.
Hmm, what I want as engineer/electrician? Normally I know which voltage will be here, but not always. In first step I want see just the signal and then I want to change grid/resolution/timebase based on it.

I see. There are two settings (as you said) one for device and one for output (recomputed) signal which can be different.

davidgraeff, 2017-08-26:
The codebase is ancient guys, before you thing about implementing skins, the application should be converted to use UI files (already deprecated) or QML.

OliverHaag, 2017-08-26:
Well UI files were there back when I started working on OpenHantek but are
just unflexible and unnecessarily complex imo. So if you want to port the
UI, port it to QML.

dendvz, 2017-08-26:
@davidgraeff QML migration attempt [mr_split_gui_backend](https://github.com/OpenHantek/openhantek/tree/mr_split_gui_backend) looks abandoned, having an unresolved dependency on [qtquickplot](https://gitlab.com/qtquickplot/qtquickplot). Any chance of resurrection?

davidgraeff, 2017-08-26:
Yes it is abandoned. I tried to do too many things at once.

dipusone, 2017-09-13:
I've also implemented a slider for selecting the probe. If needed i can make a PR.

davidgraeff, 2017-12-24:
A slider for the probe would be awesome indeed. I suggest a "standard" selection of 1x, 10x, 100x, read from the settings object. The user would be able to overwrite the available choices in the settings dialog (comma separated in a text field as a first easy implementation, a list view with +/- buttons in a more advanced implementation)

dipusone, 2017-12-24:
For now I have fixed values, i can modify the settings in order to give the user the possibility to select values. Since it is the first time for me working on QT i will need a little time to bang my head on the management of the settings :P.

dendvz, 2017-12-25:
Would be great to combine 3 enhancements (#62, #74, #107) into one probe config bundle:
- The new DsoProbe class has attributes representing probe name, display units and list of available gain (V/unit) values. The json-like description of [Hantek PP-XX](http://www.hantek.com/en/ProductDetail_73.html) probe will look like `{ "name" : "Hantek PP", "units" : "V", "gain" : [1, 10] }`. Collection of DsoProbe objects should be saved in the new "probe" group in settings, **Hantek PP** is the only item known by default.
- Each physical channel will get 2 new GUI controls: probe selection pushbutton and gain selection slider (disabled if probe has only one gain value). These values are stored in channel settings.
- #107 can be easily re-implemented by probe w/ negative gain.

dipusone, 2018-01-05:
I've finished to implement the logic and setting but now i see that the code and structure has changed from my fork. I will have to sync with master and modify some implementation and this could take some time.

davidgraeff, 2018-01-05:
@dipusone The codebase is in the process of being restructured at the moment. I'm sorry for that. There is one more commit in the queue, implementing the MainWindow in a more modern way.

dipusone, 2018-01-05:
No problem. I will wait until the codebase is restructured and i will try do adapt my code to suit!

davidgraeff, 2018-01-06:
The code base is ready now. I don't expect bigger changes anymore.

dipusone, 2018-01-14:
Sorry for the delay but Work called back from holiday! I've finished to "port" my change to the new codebase and all should be working. I can make a PR now but there are few things:
- probe settings are linked to the voltage channel
- there is no ProbeClass
- I've to document some methods :P

If you can wait in a week or so I should be able to implement the changes and optimize more the code.

davidgraeff, 2018-01-14:
Make the Pull Request early. You can always add more commits. Add a ToDo list to the PR message.
This way others see that you are working on this feature and can even comment on your code.

dipusone, 2018-01-14:
I'm officially lost. Where the data acquisition and processing has gone? Normally i would apply the probe gain the the raw data in order to reflect the change in all the other analyses, but with the new changes (the one in the last days). I cannot find where the raw data is processed and how to set/access the settings from the gui.

davidgraeff, 2018-01-14:
The raw data is passed to the `post/postprocessing` class. You can find the connection in the main.cpp. `PostProcessing` works in a different thread in a vulcano like style. `Processors` can register themself and get/modify/pass the data to the next `Processor`.
If you want to modify the voltage levels for all later processors, you should register your own `Processor` very early in the chain.
Your processor can have access to the settings object and the device specification.

closed 2018-03-09 by davidgraeff