OpenHantek/openhantek #63: Building openhantek on Mac OS 10.12.4 fails.

sbfreddie, 2017-05-01:
Hi:
I'm trying to build openhantek on Mac OS 10.12.4, and the build fails.  Here is the command line output:

`freddiecBookPro:build freddie$ cmake ../ -DCMAKE_INSTALL_PREFIX=/Users/freddie/Desktop/OpenHantek/build/bin -DCMAKE_PREFIX_PATH=/usr/local/Cellar/qt/5.8.0_1
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/freddie/Desktop/OpenHantek
freddiecBookPro:build freddie$ make -j4
[  4%] Automatic MOC, UIC and RCC for target OpenHantek
[  4%] Built target OpenHantek_autogen
[ 20%] Building CXX object openhantek/CMakeFiles/OpenHantek.dir/src/helper.cpp.o
[ 20%] Building CXX object openhantek/CMakeFiles/OpenHantek.dir/src/hantek/device.cpp.o
[ 20%] Building CXX object openhantek/CMakeFiles/OpenHantek.dir/src/dataanalyzer.cpp.o
[ 20%] Building CXX object openhantek/CMakeFiles/OpenHantek.dir/src/hantek/control.cpp.o
/Users/freddie/Desktop/OpenHantek/openhantek/src/dataanalyzer.cpp:29:10: fatal error: 'fftw3.h' file not found
#include <fftw3.h>
         ^
/Users/freddie/Desktop/OpenHantek/openhantek/src/helper.cpp:30:10: fatal error: 'libusb-1.0/libusb.h' file not found
#include <libusb-1.0/libusb.h>
         ^
1 error generated.
make[2]: *** [openhantek/CMakeFiles/OpenHantek.dir/src/helper.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /Users/freddie/Desktop/OpenHantek/openhantek/src/hantek/device.cpp:28:
/Users/freddie/Desktop/OpenHantek/openhantek/src/hantek/device.h:30:10: fatal error: 'libusb-1.0/libusb.h' file not found
#include <libusb-1.0/libusb.h>
         ^
1 error generated.
make[2]: *** [openhantek/CMakeFiles/OpenHantek.dir/src/hantek/device.cpp.o] Error 1
In file included from /Users/freddie/Desktop/OpenHantek/openhantek/src/hantek/control.cpp:36:
/Users/freddie/Desktop/OpenHantek/openhantek/src/hantek/device.h:30:10: fatal error: 'libusb-1.0/libusb.h' file not found
#include <libusb-1.0/libusb.h>
         ^
1 error generated.
make[2]: *** [openhantek/CMakeFiles/OpenHantek.dir/src/hantek/control.cpp.o] Error 1
1 error generated.
make[2]: *** [openhantek/CMakeFiles/OpenHantek.dir/src/dataanalyzer.cpp.o] Error 1
make[1]: *** [openhantek/CMakeFiles/OpenHantek.dir/all] Error 2
make: *** [all] Error 2
`
Anyone have a idea what I am doing wrong?

Thanks,
Ed

daybyter, 2017-05-01:
libusb 1.0 missing?

http://stackoverflow.com/questions/3853634/how-to-set-up-libusb-on-mac-os-x

fftw3:

http://www.fftw.org/install/mac.html

Maybe this helps. I don't have a mac, so I don't really know.

sbfreddie, 2017-05-02:
Hi:
libusb1.0 is installed, the dylibs are in the /usr/local where they can be easily found, fftw3 is also installed in the same place.  Perhaps the problem is that it is looking for static libs and not dynamic libs.  I'm not really sure what it is looking for.  I tried use all sorts of flags during the configure stage but nothing seems to work.

Thanks,
Ed

daybyter, 2017-05-02:
so the libusb file should be in 

/usr/local/libusb-1.0/libusb.h

then? What were your -I (include) flags for the compiler? Was /usr/local included?
It's often just /usr/include or /usr/local/include

You might try to set a link to a directoy, that is searched for includes.
So you link libusb.h to /usr/include/libusb-1.0/libusb.h 
, or so.

At the compile stage, static or dynamic shouldn't matter. That might cause a problem later during linking.

sbfreddie, 2017-05-02:
Hi Again:
So I tried telling Cmake where to find the libs & include files like this:
`cmake ../ -DCMAKE_INSTALL_PREFIX=/Users/freddie/Desktop/DSO-5200A-USB_Stuff/openhantek/build/bin -DCMAKE_PREFIX_PATH=/usr/local/Cellar/qt/5.8.0_1 -DCMAKE_INCLUDE_PATH=/usr/local/include -DCMAKE_LIBRARY_PATH=/usr/local/lib`

Then I told make to include the "include" files in usr/local like this:
`make -j4 -I /usr/local/include`

This seems to have helped make find the libusb stuff, but it still cannot find "fftw3.h" which is located in /usr/local/include, here is the output from the locate command:
`locate fftw3.h
/usr/local/Cellar/fftw/3.3.4_1/include/fftw3.h
/usr/local/Cellar/fftw/3.3.5/include/fftw3.h
/usr/local/Cellar/fftw/3.3.6-pl2/include/fftw3.h
/usr/local/include/fftw3.h
`
I do not know how to preceed to get the compiler to find the fftw3.h file.
Any suggestions?

Here is the whole output from the terminal window:
`freddie$ cmake ../ -DCMAKE_INSTALL_PREFIX=/Users/freddie/Desktop/DSO-5200A-USB_Stuff/openhantek/build/bin -DCMAKE_PREFIX_PATH=/usr/local/Cellar/qt/5.8.0_1 -DCMAKE_INCLUDE_PATH=/usr/local/include -DCMAKE_LIBRARY_PATH=/usr/local/lib
-- The C compiler identification is AppleClang 8.1.0.8020042
-- The CXX compiler identification is AppleClang 8.1.0.8020042
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenGL: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks/OpenGL.framework  
-- Found libusb: /usr/local/lib/libusb-1.0.dylib  
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- Found FFTW: /usr/local/lib/libfftw3.dylib  
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/freddie/Desktop/DSO-5200A-USB_Stuff/openhantek/build
freddiecBookPro:build freddie$ make -j4 -I /usr/local/include
Scanning dependencies of target OpenHantek_autogen
[  4%] Automatic MOC, UIC and RCC for target OpenHantek
Generating MOC source OpenHantek_autogen/UVLADIE3JM/moc_colorbox.cpp
Generating MOC source OpenHantek_autogen/UVLADIE3JM/moc_configdialog.cpp
Generating MOC source OpenHantek_autogen/UVLADIE3JM/moc_configpages.cpp
Generating MOC source OpenHantek_autogen/UVLADIE3JM/moc_dataanalyzer.cpp
Generating MOC source OpenHantek_autogen/UVLADIE3JM/moc_dockwindows.cpp
Generating MOC source OpenHantek_autogen/UVLADIE3JM/moc_dsocontrol.cpp
Generating MOC source OpenHantek_autogen/UVLADIE3JM/moc_dsowidget.cpp
Generating MOC source OpenHantek_autogen/UVLADIE3JM/moc_exporter.cpp
Generating MOC source OpenHantek_autogen/UVLADIE3JM/moc_glgenerator.cpp
Generating MOC source OpenHantek_autogen/UVLADIE3JM/moc_glscope.cpp
Generating MOC source OpenHantek_autogen/QCBJVIA7PY/moc_control.cpp
Generating MOC source OpenHantek_autogen/QCBJVIA7PY/moc_device.cpp
Generating MOC source OpenHantek_autogen/UVLADIE3JM/moc_levelslider.cpp
Generating MOC source OpenHantek_autogen/UVLADIE3JM/moc_openhantek.cpp
Generating MOC source OpenHantek_autogen/UVLADIE3JM/moc_settings.cpp
Generating MOC source OpenHantek_autogen/UVLADIE3JM/moc_sispinbox.cpp
Generating MOC compilation OpenHantek_autogen/moc_compilation.cpp
Generating RCC source OpenHantek_autogen/PNK5WDWK6L/qrc_application.cpp
Generating RCC source OpenHantek_autogen/PNK5WDWK6L/qrc_configdialog.cpp
[  4%] Built target OpenHantek_autogen
Scanning dependencies of target OpenHantek
[ 12%] Building CXX object openhantek/CMakeFiles/OpenHantek.dir/src/configdialog.cpp.o
[ 12%] Building CXX object openhantek/CMakeFiles/OpenHantek.dir/src/colorbox.cpp.o
[... 20 lines cut ...]
In file included from /Users/freddie/Desktop/DSO-5200A-USB_Stuff/openhantek/openhantek/src/configpages.cpp:33:
In file included from /Users/freddie/Desktop/DSO-5200A-USB_Stuff/openhantek/openhantek/src/configpages.h:31:
In file included from /Users/freddie/Desktop/DSO-5200A-USB_Stuff/openhantek/openhantek/src/dsowidget.h:32:
/Users/freddie/Desktop/DSO-5200A-USB_Stuff/openhantek/openhantek/src/levelslider.h:75:7: warning: 'LevelSlider::setVisible' hides overloaded virtual
      function [-Woverloaded-virtual]
  int setVisible(int index, bool visible);
      ^
/usr/local/Cellar/qt/5.8.0_1/lib/QtWidgets.framework/Headers/qwidget.h:470:18: note: hidden overloaded virtual function 'QWidget::setVisible' declared
      here: different number of parameters (1 vs 2)
    virtual void setVisible(bool visible);
                 ^
1 warning generated.
1 warning generated.
make[1]: *** [openhantek/CMakeFiles/OpenHantek.dir/all] Error 2
make: *** [all] Error 2
freddiecBookPro:build freddie$ locate fftw3.h
/usr/local/Cellar/fftw/3.3.4_1/include/fftw3.h
/usr/local/Cellar/fftw/3.3.5/include/fftw3.h
/usr/local/Cellar/fftw/3.3.6-pl2/include/fftw3.h
/usr/local/include/fftw3.h
freddiecBookPro:build freddie$ 
`

Thanks,
Ed

daybyter, 2017-05-02:
So fftw3.h is already in /usr/local? Hmmh...

What does

ls -la /usr/local/include/fftw3.h

show? Is this actually a file or just a softlink?

sbfreddie, 2017-05-02:
Hi Again:
Here is the output you requested:
`freddie$ ls -la /usr/local/include/fftw3.h
lrwxr-xr-x  1 freddie  admin  40 Apr  2 14:46 /usr/local/include/fftw3.h -> ../Cellar/fftw/3.3.6-pl2/include/fftw3.h`

It appears to be a softlink.

Thanks
Ed

daybyter, 2017-05-02:
So I would either add the actual path to the include options of the compiler or just copy the include file to /usr/local/include

sbfreddie, 2017-05-02:
Hi Again:
Well I tried including the path to fftw3.h which is ( /usr/local/Cellar/fftw/3.3.6-pl2/include) in the make line, but still got the same error.  So I tried adding the line: -DCMAKE_INCLUDE_PATH=/usr/local/Cellar/fftw/3.3.6-pl2/include to the Cmake configure line, but still the compiler cannot find fftw3.

Thanks,
Ed

sbfreddie, 2017-05-02:
I tried copying the fftw3.h file to /usr/local/include but the build still fails with the same error.

Thanks,
Ed

daybyter, 2017-05-03:
Did you overwrite the link? What does 

ls -la /usr/local/include/fftw3.h

say now?

Did you add the -I /usr/local/include option again?

sbfreddie, 2017-05-03:
Hi Again:
I have been very busy.  Since I could not get this build working I decided to rewrite the two cmake Find files:  FindFFTW.cmake, and Findlibusb.cmake.  After digging around the net with Google, I found some files that sort of did what I wanted, but I had to modify them quite a lot.
Now the cmake configure command (cmake ../ -DCMAKE_INSTALL_PREFIX=/Users/freddie/Desktop/DSO-5200A-USB_Stuff/openhantek/build/bin -DCMAKE_PREFIX_PATH=/usr/local/Cellar/qt/5.8.0_1) produces this output:
`-- The C compiler identification is AppleClang 8.1.0.8020042
-- The CXX compiler identification is AppleClang 8.1.0.8020042
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenGL: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks/OpenGL.framework  
-- Found libusb-1.0:
--  - Includes: /usr/local/include/libusb-1.0
--  - Libraries: /usr/local/lib/libusb-1.0.dylib
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- Found libfftw3:
--  - Includes: /usr/local/include
--  - Libraries: /usr/local/lib/libfftw3.dylib
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/freddie/Desktop/DSO-5200A-USB_Stuff/openhantek/build`

The make build was still failing with the errors that 3 files could not find either libusb.h or fftw3.h.
I still don't know why these files: dataanalyzer.cpp, device.h, and helper.cpp could not find these header files.  So I changed these 3 files to add /usr/local/include in front of the .h file preprocessor commands.
Now the project builds but the linking fails with these errors:
`[100%] Linking CXX executable OpenHantek
Undefined symbols for architecture x86_64:
  "_libusb_bulk_transfer", referenced from:
      Hantek::Device::bulkTransfer(unsigned char, unsigned char*, unsigned int, int, unsigned int) in device.cpp.o
  "_libusb_claim_interface", referenced from:
      Hantek::Device::search() in device.cpp.o
  "_libusb_close", referenced from:
      Hantek::Device::~Device() in device.cpp.o
      Hantek::Device::disconnect() in device.cpp.o
      Hantek::Device::search() in device.cpp.o
      Hantek::Device::bulkTransfer(unsigned char, unsigned char*, unsigned int, int, unsigned int) in device.cpp.o
      Hantek::Device::controlTransfer(unsigned char, unsigned char, unsigned char*, unsigned int, int, int, int) in device.cpp.o
  "_libusb_control_transfer", referenced from:
      Hantek::Device::controlTransfer(unsigned char, unsigned char, unsigned char*, unsigned int, int, int, int) in device.cpp.o
  "_libusb_free_config_descriptor", referenced from:
      Hantek::Device::search() in device.cpp.o
  "_libusb_free_device_list", referenced from:
      Hantek::Device::search() in device.cpp.o
  "_libusb_get_bus_number", referenced from:
      Hantek::Device::search() in device.cpp.o
  "_libusb_get_config_descriptor", referenced from:
      Hantek::Device::search() in device.cpp.o
  "_libusb_get_device_address", referenced from:
      Hantek::Device::search() in device.cpp.o
  "_libusb_get_device_descriptor", referenced from:
      Hantek::Device::search() in device.cpp.o
  "_libusb_get_device_list", referenced from:
      Hantek::Device::search() in device.cpp.o
  "_libusb_init", referenced from:
      Hantek::Device::Device(QObject*) in device.cpp.o
  "_libusb_open", referenced from:
      Hantek::Device::search() in device.cpp.o
  "_libusb_release_interface", referenced from:
      Hantek::Device::~Device() in device.cpp.o
      Hantek::Device::disconnect() in device.cpp.o
      Hantek::Device::bulkTransfer(unsigned char, unsigned char*, unsigned int, int, unsigned int) in device.cpp.o
      Hantek::Device::controlTransfer(unsigned char, unsigned char, unsigned char*, unsigned int, int, int, int) in device.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [openhantek/OpenHantek] Error 1
make[1]: *** [openhantek/CMakeFiles/OpenHantek.dir/all] Error 2
make: *** [all] Error 2`

I have no idea what this problem is, as I think my .dylibs are all x86_64 files.

Any ideas?

Thanks,
Ed

daybyter, 2017-05-04:
Hmmh....I would reinstall libinstall libusb-1.0 for my architecture at this point. Then check, if the error goes away.

sbfreddie, 2017-05-05:
Hi Again:
That was the first thing I did, however after rebuilding I still get the same errors.
At this point I am out of ideas.
Any other ideas would be most appreciated.

Thanks,
Ed

sbfreddie, 2017-05-05:
Hello Again:
Well I finally figured out why it was not finding the include files and the libs.  Instead of trying to explaining, I just changed 3 cmake files:
Inside the folder /openhantek/openhantek I changed the CMakeLists.txt file to make sure the FFTW and libusb includes and libs were being used during the build and linking process.

Inside the folder /openhantek/cmake I modified the two files, FindFFTW.cmake, and FindLIBUSB_1.cmake to find and set all the appropiate libs and includes and pass them to the CMakeLists.txt inside the /openhantek/openhantek folder.

Feel free to do whatever you like with them, they work for me on OSX 10.12.4.

They only problem is that the app cannot find the DSO-5200A on the USB port even though the DSO-5200A is attached when I run the app, and the dso5200ax86-firmware.hex and dso5200ax86-loader.hex are in the firmware folders.

Thanks,
Ed
PS attached are the 3 files:

[CMakeLists.txt](https://github.com/OpenHantek/openhantek/files/980296/CMakeLists.txt)

sbfreddie, 2017-05-05:
Hi Again:
I guess it does not take more than 1 files at a time, here is another one:

[FindFFTW.cmake.txt](https://github.com/OpenHantek/openhantek/files/980303/FindFFTW.cmake.txt)
[FindLIBUSB_1.cmake.txt](https://github.com/OpenHantek/openhantek/files/980304/FindLIBUSB_1.cmake.txt)

Remove the .txt and they should be good.

Thanks,
Ed

daybyter, 2017-05-11:
Does your scope need a firmware? Do you have USB rules to call fxload to load the firmware?

sbfreddie, 2017-05-11:
Hi again:
Remember I am on Mac OSX 10.12.4. 
So the instructions for Linux are not applicable.
I downloaded and built "fxload-osx" and used the driver files from the download of openhantek.
The fxload-osx shell script worked fine, and loaded the necessary firmware into the the DSO-5200A.
Openhantek sort of works, but I cannot get it to work correctly at all, the triggering does not work, and the vertical scales do not work correctly, all these problems using the built-in 2 Vpp calibrator.
I have resorted to using the program that came with the DSO-5200A on Windows under VMWare Fusion and the DSO works just as it should.
Thanks for all you help,
Ed

stkyoung, 2017-05-20:
I had the same problems during compilation, but it works for me now. I had to install the Xcode command line tools first

sbfreddie, 2017-05-20:
Hi stkyound:
Thanks, but I always install the Xcode command line tools, as many of my other projects need them.  I use embedXcode (using the Teensy's) for all of my Arduino projects.

Thanks,
Ed

commit 1436e5db (2017-12-11): Fixes macOS build on some systems: Fixes #63
closed 2017-12-11 by davidgraeff