OpenHantek/openhantek #65: 6022BE - Wrong voltage level - Issues with changing trigger mode

rmelizalde, 2017-07-14:
Hello,

First of all, sorry if I'm reporting a know issue or if I've missed any step.

I'm following the steps in README.

After loading the firmware (by udev plugdev), Openhantek is apparently working fine. Both probes are meassuring.

However, it seems the maximum voltage level I can meassure is around 2.5Volts (I've checked with a 5VDC power supply and I'm pretty confident it's around 5VDC; I have confirmed its voltage level with a multimeter).

The scope is working fine, because it works fine with the Windows software.

Additionnally, with trigger in AUTO mode, it's apparently working fine. But when in normal mode or single, going back to AUTO does not work.

I can reproduce this issue with both the tar.gz packange in openhantek.org. And with the latest source in master (commit 0ce9ae3a5619e146db1004df629d0bc9c928687d).

Additionnally, I've tried to generate the firmware hex files from the .sys drivers in the CD-ROM by compiling the extract-fw tool. but I get:
Can't get firmware contents: Bad value

Thanks a lot in advance.

Best regards,
Ricardo

davidgraeff, 2017-12-20:
@rmelizalde You listed a lot of issues in one report.

* extract-fw: This is intended to work with the website driver packages. The cdrom drivers may be too old.
* Can you please retest if the trigger works now as expected?
* You forgot to mention your device model. If you see 2.5V only if it's in reality 5V, then the gainsteps for your model are wrong and need to be fixed.

davidgraeff, 2018-01-06:
I thing a lot of mentioned issues are fixed now. Please reopen if anything still does not work for you.

closed 2018-01-06 by davidgraeff