OpenHantek/openhantek #71: Threading problems

ta33ers, 2017-08-14:
Linux Debian error message:
"QObject: Cannot create children for a parent that is in a different thread.
(Parent is Hantek::Control(0x55649f2b7600), parent's thread is QThread(0x55649f0acfe0), current thread is Hantek::Control(0x55649f2b7600)"

I followed all the instructions, am I missing something? I am having the same problem on Arch Linux, and Debian.

davidgraeff, 2017-08-14:
OpenHantek was tested with Qt5.1, newer Qt versions might be more restrictive with their thread handling. It is probably not your fault, but at the moment nobody has time to look into this.

ta33ers, 2017-08-14:
Thanks for your reply, I have got it to work, some what, I hope to try
later with a signal generator.

closed 2017-12-10 by davidgraeff