OpenHantek/openhantek #76: No frequency measurement

kulkarniniraj, 2017-09-06:
I am analyzing a PWM output filtered through RC filter. Resulting sine wave of around 1.4 KHz have poor quality, but still a reasonable shape. On enabling spectrum, I can see a distinct peak at around 1.4 KHz, but the output always shows 0 Hz. Attached screen-shot
[image]
.

dendvz, 2017-11-23:
Probably I do know the answer: DataAnalyzer::run() fails to detect the 1st peak of autocorrelation function if mean value is not subtracted from data samples before FFT.
Your screenshot says that CH1 has AC coupling, but in fact Hantek 6022 does not support it. There is a 0Hz bias seen on frequency plot, hence it is DC-coupled. Just insert a capacitor (at least 1uF) between PWM output and the probe.

Screenshot below demonstrates correlator failure (CH1 is biased - frequency measurement failed, for CH2 the capacitor is inserted - frequency measurement succeeded).
[image]

davidgraeff, 2017-11-27:
I wonder if there should be an FAQ section about OpenHanteks limitations, linked from main readme.

dendvz, 2017-12-25:
@davidgraeff I believe you can remove the labels and close issue as fixed. PRs #123 and #125 also improved frequency measurement. Measurements with 6022BE built-in generator look promising, I have to resurrect my generator and capture new screenshot for final proof. Stay tuned!

dendvz, 2017-12-26:
Nope, the frequency measurement still does not work for the ~80% duty cycle square wave. For now, documentation solution looks appropriate. Frequency measurement depends on:
- presence of DC bias in a signal
- window function chosen for spectrum analysis