OpenHantek/openhantek #77: Build instructions update for Mac OS X Sierra

zambetti, 2017-09-13:
I'm running Mac OS X 10.12.5, and there have been some changes to the environment vars that cause the build to have trouble finding libusb and fitw.

Here is what worked for me...

Added this line under line 12 of the CMakeLists.txt:
`include_directories(/usr/local/include)
`
```
mkdir build
cd build
export DYLD_LIBRARY_PATH=/usr/local/lib
export LDFLAGS="-L/usr/local/lib"
export CFLAGS="-I/usr/local/include"
cmake ../ -DCMAKE_PREFIX_PATH=/usr/local/Cellar/qt/5.9.1
make -j
```

diogoviannaaraujo, 2017-11-10:
Will build on high sierra and i guess ill need this, really sad this looks to have no more management, was hoping to a pre built binary just to be frustated by a empty releases page

davidgraeff, 2017-11-14:
I happily accept a MacOS build for the releases page. None of the contributors so far had MacOS so nobody could release and test so far.

commit 94b4c26b (2017-11-27): Update CMakeLists.txt

Fixes #77
closed 2017-11-27 by davidgraeff