OpenHantek/openhantek #78: USB port not opening until Virtualbox has been started

kschoelpple, 2017-09-18:
Hi,
first of all have to say thank you for providing this great software.  ;-))
A little problem I have encountered and may have to do with my particular installation: 
When I connect the Hantek device I can not get any signal from it and hence the LED won't come on at all.
However, when I start Virtualbox and run Windows7, then activate the Hantek USB selection then immediately the LED will come on. I can then shut down the guest system and Virtualbox and successfully operate OpenHantek.
I suspect Virtualbox is initially "hijacking" the USB port somehow ? 
Any advice / ideas how to give Ubuntu priority of the port so I won't have to start Virtualbox every time I want to use OpenHantek ?
 Thanks in advance.

**Equipment:**
Hantek 6022BE
Ubuntu 16.04
Virtualbox installed - running various guest system at times, but not always

davidgraeff, 2017-11-27:
This is not necessarily virtualbox, but probably the hantek driver on your windows 7 system, that initializes the scope. Every time you switch off/on the scope, it need to be provided with a firmware. The windows 7 driver does this. Our udev rules do this as well, if configured correctly. Should be mentioned in the readme.

davidgraeff, 2017-12-11:
Should work, now that OpenHantek transfers the firmware

closed 2017-12-11 by davidgraeff