OpenHantek/openhantek #81: 100 MS/s doesn't work on the DSO-2250

pierstitus, 2017-10-23:
On the DSO-2250 the 100 MS/s mode doesn't work, there are no updates until it is switched to a higher or lower value.
If I manually enter values this happens between 51 and 100 MS/s, from 101 on it works again. Then it shows 200 MS/s in the top display (in single channel mode).

At 125 MS/s it seems to work, when you change the max samplerate in control.cpp `   this->specification.samplerate.single.max = 125e6;`

I'm using git rev 5913ecb4776e7d3c70 (Aug 9 2017) on Ubuntu 16.04

jav, 2018-04-06:
I'm seeing a similar symptom (I havn't tried the 125MS/s change in control.cpp).

But, as I change time ranges the sample rate will jump back to 100MS/s.

I'm using 0eff8d451180c8c1328ff0d33de4e2f477147e16 on Ubuntu 18.04

Connected to this bug (I think), I also don't get any record-length options, unless I fiddle around with sample-rate and time-range.
I havn't noticed the pattern yet, but the function  `int addRecordLength(QComboBox *recordLengthComboBox, unsigned recordLength)` isn't executed until I hit some magic lower sample-rate & lower time-range.

truher, 2018-07-18:
I see this issue too.  the record length field is blank, and as a result i think it's receiving a tiny amount of data -- i can see it by magnifying the timebase to 1us.  fiddling with the sample rate makes the record length work again, and i can make everything behave normally after that.  i'm using a DSO-2090 on ubuntu 18.04.

davidgraeff, 2018-07-18:
Thanks for the report, guys. This is caused by a commit that introduced support for the DSO 6022, which does not have separate commands for sample-rate and time-range. Because I only own a DSO 6022, I cannot fix this problem right now. If anybody want to step in, please do so. Thanks.

truher, 2018-07-18:
I'd be happy to try to fix it.  Was the change in a single commit?  Does
the 6022 have adjustable record length?  I can't tell what updates the
record length widget.

jav, 2018-07-18:
I have a DSO-2250. I'll gladly help out things out.

To my best knowledge, the record length-widget is updated by:
[src/main.cpp : 68-75](https://github.com/OpenHantek/openhantek/blob/57e0bebcc1d4cf99d70071eae48604149332abd3/openhantek/src/main.cpp#L68)

and in [src/dock/HorizontalDock.cpp : 122](https://github.com/OpenHantek/openhantek/blob/master/openhantek/src/docks/HorizontalDock.cpp#L122)

(to detect what was the cause was on my end, I had added debug-printouts at both these locations, to see when/why the dropdown would be updated)

uunicorn, 2018-07-22:
+1, same issue on DSO 2250.

uunicorn, 2018-07-22:
The PR above deals with 100MS/s issue alone. It is not trying to fix the record length fun.

uunicorn, 2018-07-23:
Re empty record length dropdown:

The drop down is populated on  availableRecordLengthsChanged signal.
Currently this signal is only fired when switching between fast and normal modes.
Originally this signal was fired when sampling was enabled to initialize the widgets, but this code was commented out in 1c7190e7. Not sure what was the reasoning there, so I don't really want to touch it.

On DSO 2250 there is an easy workaround:
- start the app
- select a single channel
- go down to 4 us timebase
- select 20.5kS record length
Now if you quit the app normally, it will save your record length setting. The next time you start the app it will be using you last record length setting, even though there will be nothing in the dropdown (until you switch between the fast/normal mode again). 

Personally, I find the whole idea of having 3+ co-dependent user inputs a little bit confusing. I think it would be much easier for the user to have a predefined set of supported timebase values in a single dropdown. Fast mode should be selected automatically when timebase demands it and there is only one channel currently selected. As a user I don't care what is the current mode, record length and even sample rate. I want to trust the tool to make the best decision for me.

davidgraeff, 2018-07-24:
@uunicorn I very much support the idea of a simpler user interface. The code is unfortunately quite fragile and as long as there are no integration tests with fake usb data for every supported device, I find it difficult to even accept PRs that alter the current behaviour.

PR #247 (2018-07-24): Fix the [no] downsampler border case for DSO 2250.
downsampler == 1 means divide by one => no downsampling, use the base clocks.

On 2250, frequency divider is a 16-bit up-counter. It outputs a pulse on
overflow when enabled and reloads the value which was passed in
the ESETTRIGGERORSAMPLERATE command.

If you enable downsampler with ESETTRIGGERORSAMPLERATE and:
 - write 0xffff as a counter (0x10001 - 2), it will divide by 2
 - write 0xfffe as a counter (0x10001 - 3), it will divide by 3
 ....
 - write 0x0001 as a counter (0x10001 - 0), it will divide by 65535
 - write 0x0000 as a counter (0x1 [... cut]
commit 78623870 (2018-07-24): Fix the [no] downsampler border case for DSO 2250.

downsampler == 1 means divide by one => no downsampling, use the base clocks.

On 2250, frequency divider is a 16-bit up-counter. It outputs a pulse on
overflow when enabled and reloads the value which was passed in
the ESETTRIGGERORSAMPLERATE command.

If you enable downsampler with ESETTRIGGERORSAMPLERATE and:
 - write 0xffff as a counter (0x10001 - 2), it will divide by 2
 - write 0xfffe as a counter (0x10001 - 3), it will divide by 3
 ....
 - write 0x0001 as a counter (0x10001 - 0), it will divide by 65535
 - write 0x0000 as a counter (0x [... cut]
closed 2018-07-24 by davidgraeff