OpenHantek/openhantek #85: Low resolution display support

alextrezvy, 2017-11-26:
Running on a laptop is a cricital function because the device doesn't have the galvanic isolation. It just burns out when measuring power lines. So, is there a way to run with 1024x600 resolution?

dendvz, 2017-11-26:
Just dock all control windows into tabs and hide toolbars. I have managed to shrink main window to 886x482:
[image]

davidgraeff, 2017-11-27:
I think as well that due to the movable and closeable dock windows, openHantek is very well suited for small resolutions.

dendvz, 2017-11-27:
@davidgraeff  Hmm, closing the issue completely might be too early. Tabbed layout is not properly saved in settings. Upon restart the tabs are displayed as left-docked.
Composed layout with tabs:
[image]
Turns into:
[image]

davidgraeff, 2017-11-27:
Alright, the layout saving is not entirely correct then. That should be fixed of course.

closed 2017-12-07 by davidgraeff