OpenHantek/openhantek #88: How to customize the inf file for device to run on MS Windows

pi-null-mezon, 2017-11-29:
Hello! Thanks fot this project!

The question is: I have manually build OpenHantek on Windows, but when I run the app it reports:
**Couldn't open device 001:007: Operation not supported or unimplemented on this platform**
Project's main page says that I am *need to download the WinUSB driver files and customize the inf file for your device yourself at the moment*. But download link is not working anymore and how I should customize INF file? Please help =)

davidgraeff, 2017-12-07:
I have added the winusb driver to the cmake directory and changed the instructions in the readme. Unfortunately I don't have a windows system available right now and can't figure out the exact instructions. 

If you can solve this on your system, it would be awesome if you share your steps.

pi-null-mezon, 2017-12-08:
Good news, everyone! )

@davidgraeff, thanks alot! Thanks to your update I am been able to figure out how to run it ) Look:

[image]

What I have done to install libusb driver for Hantek6022BE:
Step 1 -  remove Hantek6022BE's native (on CD) driver via Windows device manager
Step 2 - download *winusb driver.zip* from [openhantek resources](https://github.com/OpenHantek/openhantek/tree/master/cmake)
Step 3 - unzip archive and modify libusb_device.inf. I have modified only first section:
````
; =====================================================
; ========= START USER CONFIGURABLE SECTION ===========
; =====================================================

DeviceName = "HantekDSO6022BE"
VendorID = "VID_04B5"
ProductID = "PID_6022"
DeviceClassGUID = "{78a1c341-4539-11d3-b88d-00c04fad5171}"
; Date MUST be in MM/DD/YYYY format
Date = "08/12/2017"

; =====================================================
; ========== END USER CONFIGURABLE SECTION ============
; =====================================================
````
Step 4 - physically plug (or replug) oscilloscope into PC's. From the Windows device manager update driver for the HantekDSO6022BE by manually point location of the directory where you unzip at step 2 and where modified libusb_device.inf located

P.S. Actually I have not used project's Cmake files to build and run (I have tried but it was not working for me on MS Windows, maybe I will open another issue later). Instead I have manually created Qt project and downloaded all dependencies (FFTW and libusb) by myself

pi-null-mezon, 2017-12-08:
@davidgraeff, to make a contribution, I can make MS Windows installer for the OpenHantek. It could be placed in the [release section](https://github.com/OpenHantek/openhantek/releases) and will give acces to the project for the users not familiar with programming. What do you think?

davidgraeff, 2017-12-08:
Please open an issue if the cmake build system does not work for you :)

I will add your steps to the readme for other fellow windows users.

One installer for one particular version is nice but in the mid to long run not very helpful. The cmake buildsystem together with NSIS can theoretically create an installer executable/msi. But because I do not use windows at the moment, I haven't investigated into this solution.

pi-null-mezon, 2017-12-11:
Ok, I've understood

davidgraeff, 2017-12-14:
Because we have a more detailed readme now, I'll close this issue.

closed 2017-12-14 by davidgraeff