OpenHantek/openhantek #91: Feature request >> Add RAW measurements recorder

pi-null-mezon, 2017-12-11:
Hello. It would be great if OpenHantek will allow to save RAW measurements on HDD. I am already implement such a feature in the repo fork. May I make a pull request?

davidgraeff, 2017-12-14:
Seems like an addition that does not interfere with the rest of the code base. Would be awesome if you either could make this available or allow me to copy the DataRecorder class to the master branch. (I reorganized some files recently, a rebase might be difficult)

pi-null-mezon, 2017-12-14:
So, the DataRecorder's code resides here: 

[datarecorder.h](https://github.com/pi-null-mezon/openhantek/blob/master/openhantek/src/datarecorder.h)
[datarecorder.cpp](https://github.com/pi-null-mezon/openhantek/blob/master/openhantek/src/datarecorder.h)

As I have created this class as QObject (not QThread) descendant it is important to move it into separate thread when an instance is created ([example]https://github.com/pi-null-mezon/openhantek/blob/master/openhantek/src/openhantek.cpp#L671)