OpenHantek/openhantek #94: 2250 broken with b7a65b8a

phmarek, 2017-12-14:
Since a3c13848b49d366c07a5ec2e55d351de my 2250 doesn't work any more.

Previous to that I could get it to run by starting pulseview (libsigrok) to load the firmware, and then use `OpenHantek`; with this commit the firmware seems to get loaded (I acknowledged a messagebox), but there's no data displayed. Same settings, and the same 2V/1kHz reference.

The 2250 blinks green, so it should trigger normally; sadly the commit is quite big. 

I tried to build with `cmake -DCMAKE_BUILD_TYPE=DEBUG`, but that gave quite a few errors like
```
src/hantek/hantekdsocontrol.cpp:1772:5: error: ‘Control’ has not been declared
 int Control::stringCommand(QString command) {
```

Do you have any ideas what I could do to help you making it work again??  ;)
Is there some debug/verbosity command line switch?

Thank you!

davidgraeff, 2017-12-14:
The commit is quite big unfortunately, yes. I needed to split the `types.h/.cpp` files into smaller ones to make the usb upload functionality available.

I changed two things that could cause this:
* I changed the way how locking mutexes are used. Maybe the data thread is caught in a deadlock now.
* I changed very few method signatures. Maybe a signal doesn't connect to a slot anymore.

I will investigate

davidgraeff, 2017-12-17:
Should be fixed. Please reopen if it doesn't work for you.

closed 2017-12-17 by davidgraeff

phmarek, 2017-12-17:
Thanks; I now get a waveform displayed again. (Though it takes ~4 seconds, whereas previously it was nearly instanteneous!)

But the trigger slope (rising vs. falling) and the channel settings (V/div, DC/AC) are not changed without stopping and restarting the program. The record length dropdown is empty; previously I could select 10k or 512k here.

BTW, how about _not_ asking for a device if only one is found attached?

davidgraeff, 2017-12-17:
I confirm that the record length, the trigger slope and the trigger mode doesn't work. I prepare a fix.

For my 6022 I've got a signal instantaneous though (no delay). But the 6022 only has a software trigger. Someone with a hardware trigger device needs to look at the code again.
The channel settings work for. Spectrum as well as voltage.