OpenHantek/openhantek #95: QMainWindow::GroupedDragging

voloviq, 2017-12-16:
Hi,
I have a problem with compilation on Ubuntu Studio 16.04.

The problem is in line 44 of mainwindow.cpp

"setDockOptions(dockOptions() | QMainWindow::GroupedDragging);"

Compiler issue a problem with "QMainWindow::GroupedDragging". Says that member of GroupedDragging has not belonged to QMainWindow. When I comment mentioned line compile process successful but nothing changes on screen. Green LED blink on the oscilloscope, indicate proper work with DSO2091.

My QT5 version

QMake version 3.0
Using Qt version 5.5.1 in /usr/lib/x86_64-linux-gnu

Original message 

mainwindow.cpp:44:36: error: ‘GroupedDragging’ is not a member of ‘QMainWindow’
     setDockOptions(dockOptions() | QMainWindow::GroupedDragging);

How workaround quickly this problem?

Thanks in adv
BR
Voloviq

davidgraeff, 2017-12-16:
Group dragging seem to be too new. I will conditionally apply this, depending on the Qt version. 

And yes, with the automatic firmware upload I broke the oscillator screen. I intend to fix this. Please download a git version from about a week ago, that should work.

davidgraeff, 2017-12-17:
fixed

closed 2017-12-17 by davidgraeff