OpenHantek/openhantek #97: No waveform on screen status led blinking red/green

voloviq, 2017-12-17:
Hi,
Due to some trouble with latest version I back to commit 307fb7a28d2e527c67387464f749ece7a2474c7d.
Now everything compile well without any trouble, make install as well. After installation I issue from command line OpenHantek and gets Information in new information window that "Uploading firmware to DSO-2090: done", and on console 

stop CPU
... WROTE: 792 bytes, 10 segments, avg 79
reset CPU
2nd stage: write external memory
stop CPU
2nd stage: write on-chip memory
... WROTE: 4620 bytes, 48 segments, avg 96
reset CPU
So first without clicking OK on info window status LED blinking RED about 1[s]. After clicking OK LED start blinking RED/GREEN/RED/GREEN etc.
Unfortunately no waveform is generated on screen. I make any parameters combination but without any positive result.
I'm not sure what is happen now because console don't capture any failures, warnings, etc.

davidgraeff, 2017-12-18:
Would you please test with the latest version? 307fb7a was broken.

voloviq, 2017-12-18:
Thanks David,
now works as it should. In first moment after I executed binary LED still was red/green but after change timebase only green color LED blinks. So once again thanks a lot and one suggestion please make a git TAG to assign valid version.
Best Regards

closed 2017-12-18 by voloviq