OpenHantek/openhantek #101: RAM consumption on commit fa0087aeb0970e5483f507418019a033e58c1fa3

voloviq, 2017-12-18:
Hi David,
I check latest version commit fa0087aeb0970e5483f507418019a033e58c1fa3. Everythings works fine but after 1hour or 1 and a half Openhantek consumed all of my RAM memory I mean all 8Gbytes. I discovered that because on my desktop I have installed screenlets meter applet with ram usage option sets. Of course after close application system release more than 80% of ram.
So please check maybe some objects are not released and increase continuously.

Besides that issue, everything seems to works well.

Thanks a lot
for Your job
BR
Voloviq

voloviq, 2017-12-18:
Hi again,
below I add screen after 1,5hours more or less
[image]
RAM usage still increase.

voloviq, 2017-12-18:
Finally, waveform freezes, LED green still continuous blinking. Memory still increasing
[image]
I stop testing at this moment. When the close app, memory level go back to normal usage. When again run the app. everything works well.
BR
Voloviq

davidgraeff, 2017-12-19:
Thanks for the detailed analysis. I think it is the dataanalyser class' fault.

davidgraeff, 2017-12-19:
I have found the issue. The newly introduced `DataAnalyzerResult` class doesn't free the fast fourier transformation buffer (field `window`). This way after each processed samples batch, a new buffer is allocated.

voloviq, 2017-12-19:
Hi David,
it good to hear that You detect leakage. If You want I can test fixed version. Please inform me when You finish it. Of course, You can close this issue.
Best Regards
Voloviq

davidgraeff, 2017-12-20:
@voloviq Can you please retest? I thing I fixed the ram issue

voloviq, 2017-12-20:
Hi David,
so everything compile well, after first start firmware was not found but red LED start blinking. Next segmentation fault

mw@yoga510:~/temp/openhantek_latest/openhantek/build$ OpenHantek
terminate called after throwing an instance of 'std::runtime_error*'

I used valgrind to get failure place,

mw@yoga510:~/temp/openhantek_latest/openhantek/build$ valgrind OpenHantek
==8836== Memcheck, a memory error detector
==8836== Copyright (C) 2002-2015, and GNU GPL'd, by Julian Seward et al.
==8836== Using Valgrind-3.11.0 and LibVEX; rerun with -h for copyright info
==8836== Command: OpenHantek
==8836== 
--8836-- WARNING: Serious error when reading debug info
--8836-- When reading debug info from /lib/x86_64-linux-gnu/libusb-1.0.so.0.1.0:
--8836-- Ignoring non-Dwarf2/3/4 block in .debug_info
--8836-- WARNING: Serious error when reading debug info
--8836-- When reading debug info from /lib/x86_64-linux-gnu/libusb-1.0.so.0.1.0:
--8836-- Last block truncated in .debug_info; ignoring
--8836-- WARNING: Serious error when reading debug info
--8836-- When reading debug info from /lib/x86_64-linux-gnu/libusb-1.0.so.0.1.0:
--8836-- parse_CU_Header: is neither DWARF2 nor DWARF3 nor DWARF4
==8836== Conditional jump or move depends on uninitialised value(s)
==8836==    at 0x1258512D: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.30)
==8836==    by 0x9A1F1D3: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==8836==    by 0x9A399A5: g_signal_emit_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==8836==    by 0x9A3A554: g_signal_emit_by_name (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==8836==    by 0x9A268F7: g_object_set_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==8836==    by 0x9A26FCB: g_object_set (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==8836==    by 0x5902975: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.5.1)
==8836==    by 0x58864DC: QStyleFactory::create(QString const&) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.5.1)
==8836==    by 0x581F754: QApplication::style() (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.5.1)
==8836==    by 0x581FB1C: QApplicationPrivate::initialize() (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.5.1)
==8836==    by 0x581FB6E: QApplicationPrivate::construct() (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.5.1)
==8836==    by 0x414E5D: main (main.cpp:41)
==8836== 
==8836== Conditional jump or move depends on uninitialised value(s)
==8836==    at 0x9CDB3A8: g_utf8_offset_to_pointer (in /lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.2)
==8836==    by 0x12588724: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.30)
==8836==    by 0x12588D28: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.30)
==8836==    by 0x1258905E: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.30)
==8836==    by 0x9A1EFA4: g_closure_invoke (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==8836==    by 0x9A30AFB: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==8836==    by 0x9A39D5B: g_signal_emit_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==8836==    by 0x9A3A08E: g_signal_emit (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==8836==    by 0x12717BC3: gtk_widget_realize (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.30)
==8836==    by 0x12718ED7: gtk_widget_set_parent (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.30)
==8836==    by 0x125B7A72: gtk_fixed_put (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.30)
==8836==    by 0x9A220A6: g_cclosure_marshal_VOID__OBJECTv (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==8836== 
==8836== Conditional jump or move depends on uninitialised value(s)
==8836==    at 0x9CDB3D5: g_utf8_offset_to_pointer (in /lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.2)
==8836==    by 0x12588724: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.30)
==8836==    by 0x12588D28: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.30)
==8836==    by 0x1258905E: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.30)
==8836==    by 0x9A1EFA4: g_closure_invoke (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==8836==    by 0x9A30AFB: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==8836==    by 0x9A39D5B: g_signal_emit_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==8836==    by 0x9A3A08E: g_signal_emit (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==8836==    by 0x12717BC3: gtk_widget_realize (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.30)
==8836==    by 0x12718ED7: gtk_widget_set_parent (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.30)
==8836==    by 0x125B7A72: gtk_fixed_put (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.30)
==8836==    by 0x9A220A6: g_cclosure_marshal_VOID__OBJECTv (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==8836== 
==8836== Conditional jump or move depends on uninitialised value(s)
==8836==    at 0x43E483: valueToString(double, Unit, int) (printutils.cpp:49)
==8836==    by 0x419561: SiSpinBox::textFromValue(double) const (sispinbox.cpp:73)
==8836==    by 0x59B3DDC: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.5.1)
==8836==    by 0x591FBC3: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.5.1)
[... 122 lines cut ...]
==8836== HEAP SUMMARY:
==8836==     in use at exit: 50,565,702 bytes in 58,323 blocks
[... cut ...]
==8836== Rerun with --leak-check=full to see details of leaked memory
==8836== 
==8836== For counts of detected and suppressed errors, rerun with: -v
==8836== Use --track-origins=yes to see where uninitialised values come from
==8836== ERROR SUMMARY: 28 errors from 10 contexts (suppressed: 0 from 0)

To summary, I was unable to see mainwindow at all. Please investigate code to find a leakage.
Let me know when You find a problem.

I compiled version from commit e6c6723dd3df83fb8a9930517b5e30bbd58002f2

davidgraeff, 2017-12-21:
@voloviq I was able to verify the problem and I actually introduced the runtime exception on purpose, because the scopes record length was set, before the available record lengths were requested.
I solved it now in a different way.

voloviq, 2017-12-21:
Hi David,

now is better I mean OpenHantek start working, I can see waveform main window. Nevertheless when I connected to channel 1 to test generator 1kHz. I was unable to see the square waveform. I see only one line at bottom one line at the top and from time to time one transition. Regulation of timebase, sample rate and frequency base don't give demanded effect. I encounter two times segmentation and one time 

mw@yoga510:~/temp/openhantek_latest/openhantek/build$ OpenHantek
QObject::connect: Cannot queue arguments of type 'std::vector<uint>'
(Make sure 'std::vector<uint>' is registered using qRegisterMetaType().)

I think there is still something to rework.

BR 
Michal

davidgraeff, 2017-12-21:
Unfortunately I can only test with a 6022, which bypasses a lot of the code base, because it doesn't support hardware triggers or features offset commands or internal relays. I will try to find the problem.

Was it not working when the RAM consumption was high? Or did I break it with that commit?

dendvz, 2017-12-21:
The _Cannot queue arguments of type 'std::vector'_ message should be generated by this signal/slot pair:
```
    connect(dsoControl, &HantekDsoControl::availableRecordLengthsChanged, horizontalDock,
            &HorizontalDock::availableRecordLengthsChanged);
```
solution is to place
```
qRegisterMetaType<std::vector<unsigned> >();
```
into __registerDockMetaTypes()__.

I am not sure if my observations are somehow related to this issue, but since one of the latest commits  brightness of the waveform spontaneously changes... Might be a manifestation of race condition.

davidgraeff, 2017-12-21:
Race conditions shouldn't be possible with the current design. I think it might be related to a color settings override bug.

commit 600b1dcf (2017-12-22): Register std::vector<unsigned> to the Qt metadata system to make it work with signal/slots. Fixes #101
closed 2017-12-22 by davidgraeff