OpenHantek/openhantek #103: Compatibility with Raspberry Pi

WaveLicker, 2017-12-19:
Hi,

Is openhantek compatible with Raspberry Pi ? For the moment when I try to compile, it freeze the Raspberry Pi with CPU = 100%, during make -j command.

Thanks for your help and support

Best Regards

Alain

davidgraeff, 2017-12-19:
Compile with `make` only. The j flag tells make to use all processor cores * 2 for compiling and the tinny raspberry hardware will be overwhelmed. The hardware and available OpenGL version should be sufficient though. Please reopen if you encounter any problems. And report back if it works as well :)

WaveLicker, 2017-12-21:
Hi,
it dosen't freeze anymore but when I do a make, it does a lots of error at 43%. When he wants to build glscope. Did I miss some libraries for OpenGL that are not by default on Rasbian ? I just test with the actual github version. Now I can use the application on my Ubuntu Laptop without any issue, but the same version have errors on Raspbian.

Best Regards

Alain

FCPlech, 2017-12-21:
Try to use "make -j4"

davidgraeff, 2017-12-21:
A lot of errors is not very helpful here. Could you post the errors here?

WaveLicker, 2017-12-21:
> A lot of errors is not very helpful here. Could you post the errors here?

For sure, it doesn't help... But if you told me "Yes of course, it's your OpenGL..." (I was thinking it's the issue).

Unfortunatelly I forgot my Raspberry Pi at work. I will copy paste all the Terminal error message tomorrow. Is there a version that shoud work?

Thank you very much for this amazing support. I allready build it on my Ubuntu laptop and it is powerfull! My dream is to run it into a small Raspberry Pi with a 7'' Touch screen. Do you know people that have allready OpenHantek running under Rasrpberry Pi or am I the only one with this kind of idea?

Best Regards

WaveLicker, 2017-12-22:
Hi,

here are the error that I get when compile on a Raspberry Pi (withe the current GitHub version):

Thank you for your help

[ 34%] Building CXX object openhantek/CMakeFiles/OpenHantek.dir/src/glscope.cpp.o
/home/pi/openhantek-master/openhantek/src/glscope.cpp: In member function ‘virtual void GlScope::initializeGL()’:
/home/pi/openhantek-master/openhantek/src/glscope.cpp:23:18: error: ‘glPointSize’ was not declared in this scope
     glPointSize(1);
                  ^
/home/pi/openhantek-master/openhantek/src/glscope.cpp:28:18: error: ‘GL_SMOOTH’ was not declared in this scope
     glShadeModel(GL_SMOOTH /*GL_FLAT*/);
                  ^~~~~~~~~
/home/pi/openhantek-master/openhantek/src/glscope.cpp:28:39: error: ‘glShadeModel’ was not declared in this scope
     glShadeModel(GL_SMOOTH /*GL_FLAT*/);
                                       ^
/home/pi/openhantek-master/openhantek/src/glscope.cpp:29:28: error: ‘glLineStipple’ was not declared in this scope
     glLineStipple(1, 0x3333);
                            ^
/home/pi/openhantek-master/openhantek/src/glscope.cpp:31:25: error: ‘GL_VERTEX_ARRAY’ was not declared in this scope
     glEnableClientState(GL_VERTEX_ARRAY);
                         ^~~~~~~~~~~~~~~
/home/pi/openhantek-master/openhantek/src/glscope.cpp:31:40: error: ‘glEnableClientState’ was not declared in this scope
     glEnableClientState(GL_VERTEX_ARRAY);
                                        ^
/home/pi/openhantek-master/openhantek/src/glscope.cpp: In member function ‘virtual void GlScope::paintGL()’:
/home/pi/openhantek-master/openhantek/src/glscope.cpp:44:18: error: ‘GL_LINE_STIPPLE’ was not declared in this scope
         glEnable(GL_LINE_STIPPLE);
                  ^~~~~~~~~~~~~~~
/home/pi/openhantek-master/openhantek/src/glscope.cpp:46:86: error: ‘glColor4f’ was not declared in this scope
         glColor4f(trColor.redF(), trColor.greenF(), trColor.blueF(), trColor.alphaF());
                                                                                      ^
/home/pi/openhantek-master/openhantek/src/glscope.cpp:59:70: error: ‘glVertexPointer’ was not declared in this scope
             glVertexPointer(2, GL_FLOAT, 0, &vaMarker[marker].front());
                                                                      ^
/home/pi/openhantek-master/openhantek/src/glscope.cpp: In member function ‘virtual void GlScope::resizeGL(int, int)’:
/home/pi/openhantek-master/openhantek/src/glscope.cpp:76:18: error: ‘GL_PROJECTION’ was not declared in this scope
     glMatrixMode(GL_PROJECTION);
                  ^~~~~~~~~~~~~
/home/pi/openhantek-master/openhantek/src/glscope.cpp:76:31: error: ‘glMatrixMode’ was not declared in this scope
     glMatrixMode(GL_PROJECTION);
                               ^
/home/pi/openhantek-master/openhantek/src/glscope.cpp:79:20: error: ‘glLoadIdentity’ was not declared in this scope
     glLoadIdentity();
                    ^
/home/pi/openhantek-master/openhantek/src/glscope.cpp:84:16: error: ‘glOrtho’ was not declared in this scope
             1.0);
                ^
/home/pi/openhantek-master/openhantek/src/glscope.cpp:86:18: error: ‘GL_MODELVIEW’ was not declared in this scope
     glMatrixMode(GL_MODELVIEW);
                  ^~~~~~~~~~~~
/home/pi/openhantek-master/openhantek/src/glscope.cpp: In member function ‘void GlScope::drawGrid()’:
/home/pi/openhantek-master/openhantek/src/glscope.cpp:95:15: error: ‘GL_POINT_SMOOTH’ was not declared in this scope
     glDisable(GL_POINT_SMOOTH);
               ^~~~~~~~~~~~~~~
/home/pi/openhantek-master/openhantek/src/glscope.cpp:96:15: error: ‘GL_LINE_SMOOTH’ was not declared in this scope
     glDisable(GL_LINE_SMOOTH);
               ^~~~~~~~~~~~~~
/home/pi/openhantek-master/openhantek/src/glscope.cpp:101:74: error: ‘glColor4f’ was not declared in this scope
     glColor4f(color.redF(), color.greenF(), color.blueF(), color.alphaF());
                                                                          ^
/home/pi/openhantek-master/openhantek/src/glscope.cpp:102:64: error: ‘glVertexPointer’ was not declared in this scope
     glVertexPointer(2, GL_FLOAT, 0, &generator->grid(0).front());
                                                                ^
/home/pi/openhantek-master/openhantek/src/glscope.cpp: In member function ‘void GlScope::drawGraphDepth(int, int, int)’:
/home/pi/openhantek-master/openhantek/src/glscope.cpp:123:82: error: ‘glColor4f’ was not declared in this scope
     glColor4f(trColor.redF(), trColor.greenF(), trColor.blueF(), trColor.alphaF());
                                                                                  ^
/home/pi/openhantek-master/openhantek/src/glscope.cpp:124:86: error: ‘glVertexPointer’ was not declared in this scope
[... 17 lines cut ...]
                     ^
/home/pi/openhantek-master/openhantek/src/glscope.cpp:141:114: error: ‘glTranslatef’ was not declared in this scope
         glTranslatef(-(settings->scope.horizontal.marker[0] + settings->scope.horizontal.marker[1]) / 2, 0.0, 0.0);
                                                                                                                  ^
/home/pi/openhantek-master/openhantek/src/glscope.cpp:157:43: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
             for (int channel = 0; channel < settings->scope.voltage.size(); ++channel) {
                                   ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
     if (this->zoomed) glPopMatrix();
                                   ^
openhantek/CMakeFiles/OpenHantek.dir/build.make:422: recipe for target 'openhantek/CMakeFiles/OpenHantek.dir/src/glscope.cpp.o' failed
make[2]: *** [openhantek/CMakeFiles/OpenHantek.dir/src/glscope.cpp.o] Error 1
CMakeFiles/Makefile2:153: recipe for target 'openhantek/CMakeFiles/OpenHantek.dir/all' failed
make[1]: *** [openhantek/CMakeFiles/OpenHantek.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

davidgraeff, 2017-12-23:
Cmake on my raspberry failed with 
```-- Could NOT find OpenGL (missing:  OPENGL_gl_LIBRARY OPENGL_INCLUDE_DIR)```
I have extended the ubuntu build instructions. Please install "mesa-common-dev" and "libgl1-mesa-dev".

Hoagie572, 2017-12-27:
Hi,

Just as a matter of interest, where did you get qt 5.4 or higher on the pi? Mine seems to be 5.3.

Thenks,
Geoff.

davidgraeff, 2017-12-27:
You need a recent operating system version. Debian after Jessi and Ubuntu 16.04 I think

Hoagie572, 2017-12-27:
I feel a bit of an updatng session coming on then.

WaveLicker, 2017-12-31:
@davidgraeff : 
> Cmake on my raspberry failed with
> -- Could NOT find OpenGL (missing: OPENGL_gl_LIBRARY OPENGL_INCLUDE_DIR)
> I have extended the ubuntu build instructions. Please install "mesa-common-dev" and "libgl1-mesa-dev".

I tried to isntall that both, but there were allready there.

Then, you said:

> You need a recent operating system version. Debian after Jessi and Ubuntu 16.04 I think

So I install Ubuntu MATE 16.04 and begin the process again. i still have the same issue as previous.

I installed all the think you mentioned on the file:
apt install g++ cmake qttools5-dev qttools5-dev-tools libfftw3-dev binutils-dev libusb-1.0-0-dev libqt5opengl5-dev

+ mesa-common-dev and libgl1-mesa-dev

Did you sucessfull install openhantek on your raspberry ? If yes, could you give me your magic trick?

thank you very much for your suport

davidgraeff, 2017-12-31:
apt-get install freeglut3 freeglut3-dev

WaveLicker, 2017-12-31:
I install freeglut3 and freeglut3-dev but still the same issue (sorry ). Did I have to also install something arround OpenGL that is not by default on Ubuntu for Raspberry Pi ?

davidgraeff, 2018-01-02:
I don't use Ubuntu on the pi and it compiles for me. Please find the correct packages yourself, this is not a help forum, it's a bug tracker :)

If there is a file gl.h it should contain all those functions that seem.to be missing for you. Maybe there are multiple gl.h files on your system and the OpenHantek buildsystem is picking up the wrong one?

jamieclements, 2018-01-03:
Hello,

First thing, happy new year!

I appreciate you want this to go away but thought I'd chip in.

I've been following along for a few weeks now on RPI3 with raspbian stretch and ubuntu mate.

With clean images of both I encountered the same make errors.

I did try change the entry pointing to gl.h in glscope.cpp or one of it's headers (sorry I can't remember exactly which one I modified as at work PC now) and the errors went went away but I got new ones related to QT5.

The RPI3 ubuntu mate install has 3 variations of folders containing GL.h in /usr/include :)

If you could let me know where to specify the GL.h to use I can try them all?

Jamie

jamieclements, 2018-01-03:
Obviously if you want to put it to bed I'll just go hacking around myself :)

davidgraeff, 2018-01-08:
I found the culprit. It depends on how Qt is compiled (if it uses OpenGL 1.2 oder OpenGL ES). Newer Qt versions are compiled for OpenGL ES, which do not support those ancient drawing calls that we use in the code. I created a new issue #146.

closed 2018-01-08 by davidgraeff

jamieclements, 2018-01-08:
Hi,

Sorry its a bit late now but I did get time to tru moving the different versions GL.h about between /usr/include and /opt/vc/include with no success. But you will know that now :)

The PI seems to use the one in /usr/include/GL

If it helps after installing all the freeglut/openmesa versions mentioned you will have GL, GLES2 and GLES3 folders in /usr/include with matching .h files.

Jamie

artaslove, 2018-02-22:
For best results with Raspberry Pi 3: 

Upgrade to debian stretch for proper qt support. 
Select "OpenGL with Fake KMS" under Advanced options in raspi-config. Reboot.
Compile. 
Launch application with "--useGLES" switch.

nomocnc, 2018-02-25:
Hi

Sorry late reply again.

after commit with modifed GL ~11th jan the build had no errors, also referenced in  #146  #151 

+1 for @artaslove settings, can confirm working a treat, thanks!

@davidgraeff thanks for the hard work on getting this sorted for Raspberry Pi 👍  

Screenshot below showing the docks allow GUI to scale on oddball 800x480 screen if documentation helps.

[image]

Jamie

Kulturnilpferd, 2018-03-14:
Works for me like a charm, thank you this great work 👍  
Just want to say that on the actual up-to-date Stretch of Raspbian announce that some GCC commands will change in GCC 7.1 or something like that while compiling.

WaveLicker, 2018-04-02:
I follow your procedure:
- enable fake GL
- compile
---useGLES

and now it works. I have a signal in my window!! :-)

@nomocnc : how do you obtain these windows as shown on screenshot? I'm really interrested. I would like to use my raspberry with a 7inch 800x480touch srcreen.

++

Alain

nomocnc, 2018-04-02:
@WaveLicker 
Hi Alain,
@davidgraeff has kindly coded the different sections to be moved with the mouse. This allows the window to be configured differently. 

You can play about with dragging the items around. They will stick togther and create "tabs" if you do it correctly. Have a play!

For mine I have added the spectrum, volt, trigger and horizontal windows to the right side of the window so they produce the "tabs" at the bottom.

I believe you can then save the view in the program settings.

I'm not near the machine but message me if you have no success. I can check if there is a config file for these settings and send it to you.

I'm actually using a 5" 800x480 over HDMI and using VNC over WiFi to control from a PC in a box with Li-ion battery etc. It's quicker than the new Rigol and quieter so very useful for most signals.

Jamie

WaveLicker, 2018-04-23:
Hi,

Thanks for all your advise. Now it works perfectly ! The tabs settings are nice.

I use the 7inch display and it is nice!

Just a suggestions for peoole who would like to do the same. CPU is at about 90% when sampling at 2MS/s. After a while, a temperature warning is blinking. To avoid this I add a dual fan with thermal dissipator and thermal tape. I'm testing now for one hour sampling and the température is stable.

Have a nice day

Alain

eitriv, 2018-06-05:
If you prefer not to install you can still start it with one touch by using a desktop icon file:

[Desktop Entry]
Type=Application
Comment=OpenHantek
Name=Scope
Exec=lxterminal -e /home/pi/openhantek-master/build/openhantek/OpenHantek --useGLES
Icon=/home/pi/openhantek-master/openhantek/res/images/openhantek.png
Terminal=true
Categories=Utility
StartupNotify=true
Name[en_US]=Scope

Just replace the links if the installation folder is different.
Thanks all for this great openhantek project.

smj005, 2018-06-22:
First of all I would like to thank @eitriv for giving such a step by step description for installing openhantek. I do followed each and every step but unfortunately I am getting the error below.

pi@raspberrypi:~/openhantek/build $ sudo make install
[  1%] Automatic moc, uic and rcc for target OpenHantek
[  1%] Built target OpenHantek_automoc
[  3%] Linking CXX executable OpenHantek
/usr/bin/ld: CMakeFiles/OpenHantek.dir/src/glscope.cpp.o: undefined reference to symbol 'glClear'
//usr/lib/arm-linux-gnueabihf/libGLESv2.so.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
openhantek/CMakeFiles/OpenHantek.dir/build.make:1713: recipe for target 'openhantek/OpenHantek' failed
make[2]: *** [openhantek/OpenHantek] Error 1
CMakeFiles/Makefile2:153: recipe for target 'openhantek/CMakeFiles/OpenHantek.dir/all' failed
make[1]: *** [openhantek/CMakeFiles/OpenHantek.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

Actually this error happened at the end of the execution of "make -j2". But then I tried the optional install method "sudo make install" and I got the same error again. So, could anyone help me out.

Thanks in advance.

artaslove, 2018-06-22:
glClear is part of glut... try:

`find /usr/include/ -name "glut.h"`

On my working system, the freeglut3 library is installed and exists in /usr/include/GL/glut.h

Maybe there was a problem installing that library? 

Hope this helps.

smj005, 2018-06-23:
Thank you @artaslove. It seems i have made a mistake while installing the libraries. So i reinstalled everything from the beginning now its working fine.