OpenHantek/openhantek #113: 6022BE firmware issue in current gitHub project

WaveLicker, 2017-12-22:
Hi,
I have an issue when connecting my 6022BE with the current builded version. I have an error message saying "DSO-6022BE : Firmware upload failed". and a second message : A device was found but the firmware upload seems to have failed or connection could notbe established".

Now, if I leave connected my 6022BE and open a previous release (18.12.17). I have no issue and have my oscilloscope running. If I close that version , keep the USB cable connected to my laptop and then open the last release. It works perftectly. If I disconnect the USB cable the application close and if I open the application again: issue come back.
So I have to open older version first.

davidgraeff, 2017-12-22:
Yes, you are right. The actual call to `UploadFirmware().startUpload` got accidentally removed in `main.cpp`.

velteyn, 2017-12-23:
we solved starting program with sudo, after we was in Super User mode the problem evaporated ;)

davidgraeff, 2017-12-23:
U need to install the udev rule. A GUI program running as root is never a good idea. The firmware upload and recognising the device immediately after is still buggy so I leave this report open for now.

commit 8d620a74 (2017-12-22): Fix firmware upload. Fixes #113
PR #122 (2017-12-24): Rework device selection
* Use own QDialog class SelectDevice and clean up main.cpp.
* USBDevice: Introduce a unique usb id for a connected device
* USBDevice: Introduce iteration field, to remember in which find round a device was found
Fixes #113
closed 2017-12-24 by davidgraeff