OpenHantek/openhantek #117: Acer Aspire One compilaton hangs (and also the pc hangs)

velteyn, 2017-12-22:
Hello all,
I'm attempting to compile the project on my ACER ASPIRE ONE N450 with Intel ATOM 2 Mb RAM....
Compilation hangs  on 97 %. I let it go on for 3 hours but at the end I gave up. All the machine hang and I had to reboot it. I have ZORIN OS 10 (an Ubuntu derivate)
I would like to use that PC because it have a very small display and it is very nice instead of the big laptop.

Can you help me ? :) thanks

KimBP, 2017-12-22:
Even my ubuntu running at an Intel i7 will hang for 5 minutes. Running e.g. make -j 6 solves the problem at my PC.
The number denotes the number of jobs to run in parallel. Relate it to the number of cores you have on your machine

velteyn, 2017-12-22:
I'll give it a try tkz

davidgraeff, 2017-12-22:
I will change the readme and recommend "make -j2". I didn't know that so many linux kernels run out of resources when limits are reached.

closed 2017-12-23 by davidgraeff

velteyn, 2017-12-23:
yess, I gave it a try and now works thank you

PR #118 (2017-12-23): Update build.md
Fixes #117. Many weak systems run out of resources and finally dead lock when compiled with "make -j"
commit a96f3c65 (2017-12-23): Update build.md

Fixes #117. Many weak systems run out of resources and finally dead lock when compiled with "make -j"