OpenHantek/openhantek #124: 6022BL - Square wave when measuring

dalva24, 2017-12-25:
Greetings,

I have built the [latest commit](https://github.com/OpenHantek/openhantek/commit/db4f8ba0b7b2bcc9ee7f41c96b597cdda56acacc) on Ubuntu 16.04, following the steps in build.md

When I start measuring, both channel only show square wave, as can be seen in the screenshot (channel 1 shown):
[image]

I tried grounding the probe tip, attaching the probe tip to the calibration output, even removing the probe altogether, but the waveform stays the same.

I have tried using the scope in windows using the Hantek software before and it works, so the hardware is supposedly fine.

Thanks

davidgraeff, 2017-12-25:
I can confirm this. I don't know if it is because of the firmware that we send to the 6022 or if a recent change is causing this. Needs to be investigated.

davidgraeff, 2018-01-04:
It is not a recent change. This behaviour is since first adding support for 6022. I guess the firmware we use does only work for 6022BE and not BL

dalva24, 2018-01-05:
I'm unable to build on Windows, so I can't try flashing using original firmware then start OpenHantek as you suggested. Is it possible to extract the firmware from the official Windows BL driver?

closed 2018-01-06 by davidgraeff

eagafonov, 2018-01-09:
I confirm: the new firmware works on 6022BL. 
Tested on Debian "Jessie".

commit f9e18268 (2018-01-06): Add proper firmware for 6022BL. Fixes #124
PR #138 (2018-01-06): 6022BL support
* Use enum for scope.horizontal.samplerateSource. Remove softwaretriggersettings, use dsotriggersettings.
* Move more comments from cpp to header
* Add proper firmware for 6022BL. Fixes #124
commit 792ab9cd (2018-01-06): Add proper firmware for 6022BL. Fixes #124