OpenHantek/openhantek #126: Add slider on commands section

velteyn, 2017-12-26:
Hello,
we are using a small Acer Aspire One here and its monitor is very small (11'') .
It would be very useful to have a vertical slider on the command bar on the right so to be able to access all the bottom commands. Also is it possible to resize the measure screen ? The exact monitor height can be deduced by the image we attach.

Thank you
[image]

dendvz, 2017-12-26:
@velteyn Can you use tab layout of dock windows? Please refer to #89, I've managed to shrink window to 800x600.

davidgraeff, 2017-12-26:
We are missing a features section in the readme. It need to be mentioned somewhere that tabs are possible I think. A vertical slider is not possible. The scope view has a minimum required size as well but as @dendvz said already, 800*600 is possible.

closed 2017-12-26 by davidgraeff

velteyn, 2017-12-26:
ok thanks