OpenHantek/openhantek #129: Hantek 6022BE Instructions

andrewcharnley, 2017-12-30:
Hi,

Can you explain in the documentation how to get the firmware extracted and uploaded?

Presently the firmware extraction tool won't work for the drivers on the website (both CD and website drivers are the same). I'm unsure how to proceed.

Regards, 
Andrew

dendvz, 2017-12-30:
Basically, you do not need to extract the firmware, it is already in the codebase. An application will upload the firmware at startup. Just follow [build instructions](https://github.com/OpenHantek/openhantek/blob/master/docs/build.md).

andrewcharnley, 2017-12-30:
OK. I get "Firmware upload failed"

```
./OpenHantek
stop CPU
... WROTE: 792 bytes, 10 segments, avg 79
reset CPU
2nd stage: write external memory
stop CPU
2nd stage: write on-chip memory
... WROTE: 5092 bytes, 68 segments, avg 74
reset CPU
```

dendvz, 2017-12-30:
Well, seems that this step is not reflected in build instructions. After
```
sudo make install
```
you should reload udev rules
```
udevadm control --reload-rules
udevadm trigger --attr-match=subsystem=usb
```
 or just reboot your PC.

andrewcharnley, 2017-12-30:
Sadly that's not it. Issue continues.

dendvz, 2017-12-30:
BTW, are you using git HEAD revision? Issue looks like #113, which is already fixed.

andrewcharnley, 2017-12-30:
No, latest tag. You are correct it is the same issue and fixed in master. Thanks!

davidgraeff, 2017-12-30:
It's sufficient to replug the device. Udev reloads modified rule files automatically and the trigger event is the actual replug.

closed 2017-12-31 by davidgraeff