OpenHantek/openhantek #134: Cmake met problem in windows

mike9755, 2018-01-01:
I got wrong message when using cmake to congifure win-project, detail as picture, can someone help me on this? thank you very much.
[image]

davidgraeff, 2018-01-02:
This is not an error, it is only a warning message. It just tells you that you can't compile the firmware extractor on windows. I will remove the warning for windows users.

closed 2018-01-02 by davidgraeff

mike9755, 2018-01-02:
Dear David 
I download the latest version code, and use cmake to build the project for windows, but still got an error, please help me , thank you.
[image]

mike9755, 2018-01-03:
Dear David
I have build the VS project successfully, but now I meet a new problem about build the project in visual studio as below picture.
[image]