OpenHantek/openhantek #135: Build error in visual studio

mike9755, 2018-01-03:
I got 4 error messages when try to build this project without any modification,detailed information as picture,please help me, thank you very much.

[image]

davidgraeff, 2018-01-03:
Which VS version? We have a continous integration solution with a windows visual studio 2015 installation, which checks every commit and pull request. Here it seems that `VERSION` is not defined. This comes from the cmake buildsystem in the file CPackInfos.cmake:
`set(CPACK_PACKAGE_VERSION "${DATE_VERSION}_${VCS_REVISION}")`
and in openhantek/CMakeLists.txt:
`add_definitions(-DVERSION="${CPACK_PACKAGE_VERSION}")`

mike9755, 2018-01-04:
Dear David
I used vs17 community to build th is project got that 4 errors , I reinstall a vs15 to build today but failed again with some other errors. thanks.

marktoml, 2018-01-04:
I have been looking at getting it going with VS17 community edition and with CMake, if I get a chance I’ll focus on it this weekend.

Mark
P.S. I am assuming it builds well elsewhere 😊

davidgraeff, 2018-01-06:
VS2017 can work with cmake projects natively. This might indeed work different and produce errors which do not occur with the "old school" way of generating VS projects with the cmake executable first.
Try with defining VERSION in main.cpp like this:
`#define VERSION "TEST"` and see if it works.

mike9755, 2018-01-07:
I'm completely confused now, I got 3700+ errors now, there three kinds of errors "LNK2001" "LNK2019" and "LNK1120", all them were showed in the picture, my cmake version is 3.10.1, QT version is 5.4.1
[image]

davidgraeff, 2018-01-07:
@mike9755 I compiles fine now though. The second step, the linking, fails for you. It looks like VS doesn't find the Qt libraries.

On Windows you usually have to set the Qt header path and Qt library path manually in the cmake GUI. I don't know how to do that with the native cmake support in VS.

mike9755, 2018-01-07:
Dear David
My cmake GUI screen as below, please help me how to set the Qt header path manually, thanks.

[image]

davidgraeff, 2018-01-07:
Ah ok. You are compiling with the visual studio compiler but you are linking to the mingw Qt libraries. They are not compatible because Microsoft introduced multiple linking standards over the years. You need to install a compatible Qt with the VS version that you have.

mike9755, 2018-01-09:
You are right, life is really hard on windows like you said, I've fixed most question now, I resinstalled a VS2015 and installed "qt-opensource-windows-x86-msvc2015-5.8.0.exe" which suit to VS, most errors have gone now, but the default fftw package is also mingw compiled, I find I can't fix this problem, can you kindly send me your windows compatible fftw package?my email is "mike97555@gmail.com",  or give new ideals to fix this problem, thank you very much
[image]
[image]

marktoml, 2018-01-09:
That is about where I stalled last session 😊
I will look into it again as soon as I get a break, but if someone has this working … clue me in.

Mark

davidgraeff, 2018-01-11:
It starts with a selection screen, always. Try to start in debug and see if it maybe crashes for you (which I actually don't believe). And make sure, you don't use the "Start" button of visual studio while "ALL_BUILD" is still the current project. You need to select "OpenHantek" as project first. CMake is really not made for VS unfortunately.

davidgraeff, 2018-01-12:
Because this is not an issue of OpenHantek I close this report now :)

closed 2018-01-12 by davidgraeff

mike9755, 2018-01-12:
Ok, thank you for your help