OpenHantek/openhantek #144: Zoom view and print/export enhancements

dendvz, 2018-01-07:
Would be great to display on the zoom view and print everywhere:
- channel/spectrum offset
- trigger level and position
- marker positions (N/A for zoom view)

The zoom view issue can be addressed by adding sliders w/ enabled=false, I can take care of this. The exporter, however, might require refactoring.

davidgraeff, 2018-01-08:
What kind of refactoring do you need for the exporter? The zoom view changes sound reasonable.

dendvz, 2018-01-08:
Several observations:
1) Both DsoWidget and Exporter classes render label texts independently. Would it make sense to introduce the new class which holds all formatted texts?
2) DsoWidget uses automatic placement via embedded layouts, Exporter layouts are hard-coded. Moreover, width and height are intermixed in one formula for no clear reason.
```
        stretchBase = (double)(paintDevice->width() - lineHeight * 10) / 4;
```
3) The grid is generated twice, can we reuse the GlGenerator::grid() for Exporter?
4) Adding markers & sliders to Exporter will also require cloning DsoWidget code.

davidgraeff, 2018-01-08:
Would be awesome to unify the drawing. But this is not that easy actually.
1. There are still legacy OpenGL functions used. That need to be converted to something modern like OpenGL Core 3.3 or OpenGL ES 2.0. I expect GlGenerator to change. For now the grid and scope drawing code needs to be duplicated until this is done.
2. I second that the layout approach of DsoWidget should be used for Exporter as well. One shared utility class should construct the layout including the texts etc.

davidgraeff, 2018-01-12:
I have finished the OpenGL3 renderer and discovered that the renderer is size independent. It should be possible to reuse that for the Exporter, where the rendering then can use the size of a Paper (print) or the export image size.
If we realize that the layout of DsoWidget is created by a helper class, we can reuse that as well, so that finally no code duplication is required anymore.