OpenHantek/openhantek #147: Can't build extractfw on OSX

DanielO, 2018-01-11:
I am having trouble with my DSO-2150 so I thought I'd check I had the latest firmware but unfortunately the extractor doesn't build.

Firstly it wants malloc.h which is deprecated so I changed that to stdlib.h and now get..
````[Maelstrom 14:53] ~/projects/openhantek/build/firmware >make
[ 50%] Building C object firmware/CMakeFiles/FirmwareExtractor.dir/extractfw.c.o
In file included from /Users/oconnd1/projects/openhantek/firmware/extractfw.c:31:
/opt/local/include/bfd.h:35:2: error: config.h must be included before this header
#error config.h must be included before this header
 ^
1 error generated.
make[2]: *** [firmware/CMakeFiles/FirmwareExtractor.dir/extractfw.c.o] Error 1
make[1]: *** [firmware/CMakeFiles/FirmwareExtractor.dir/all] Error 2
make: *** [all] Error 2
````

DanielO, 2018-01-11:
I hacked a bit more and got it building by adding...
````
#define PACKAGE_VERSION 1
#define PACKAGE 1
````
before #include <bfd.h>

After that I had to link with libintl, libz and libiberty but it doesn't like the .sys files..
````
[Maelstrom 15:04] ~/projects/openhantek/build/firmware >./FirmwareExtractor ~/Downloads/DSO2150_Driver/Dso2150x861.sys
Error opening file: Invalid bfd target
[Maelstrom 15:04] ~/projects/openhantek/build/firmware >./FirmwareExtractor ~/Downloads/DSO2150_Driver/Dso2150x862.sys
Error opening file: Invalid bfd target
````
Is there another way to extract these or is my build broken?

davidgraeff, 2018-01-11:
Because nobody had a DSO-2150 recently, it might be that the current OpenHantek build is broken for that device. Can you tell me about the exact trouble you have?

DanielO, 2018-01-11:
I'll open a separate issue for the DSO-2150.

Note that I realised I could just build extractfw in a Linux VM and get the .hex files out that way so I have a work around for this particular issue for now.

PR #151 (2018-01-12): Opengl3
Contains a lot of code base changes. The post processing directory from my old branch is finally established.

The directory dependency situation is clearly written down in readme files now. 

The OpenGL renderer was a chunk of work before it actually was able to draw anything. The solution is more clean now though. Should compile and run on Raspberry Pi or Android solutions now.

Todo:
* The new glScopeGraph class needs documentation.

Fixes #146 
Fixes #147
closed 2018-01-12 by davidgraeff