OpenHantek/openhantek #148: support for DSO-6082BE

alexanderustinov, 2018-01-11:
Hi, recently have become a user of DSO-6082BE. Would like to help with support for that model.

For now have tried:
1. Extracting firmware from windows cd with FirmwareExtractor. It fails with 
`BFD: hex/Hantek6082BEAMD642.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .text`
`BFD: hex/Hantek6082BEAMD642.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .rdata`
`BFD: hex/Hantek6082BEAMD642.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .data`
`BFD: hex/Hantek6082BEAMD642.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .pdata`
`Section .data found (starting at 0x3500, 272 bytes)`
`Symbol _firmware found (offset 0xffffffffffffcb00, 0 bytes)`
`Symbol _loader found (offset 0xffffffffffffcb00, 272 bytes)`
`Can't get firmware contents: Bad value`
2. Dumb changes of DSO6022.cpp to make it fit DSO6082.
(firmware was already loaded with virtual windows machine)
For that openhantek fails with
`Sending control command e0 failed: Pipe error`
`Sending control command e1 failed: Pipe error`
`Sending control command e2 failed: Pipe error`
`Sending control command e3 failed: Pipe error`
`Getting sample data failed:  "Pipe error"`
`Sending control command e0 failed: Pipe error`
`Sending control command e1 failed: Pipe error`
`Sending control command e2 failed: Pipe error`
`Sending control command e3 failed: Pipe error`
`Getting sample data failed:  "Pipe error"`
`terminate called after throwing an instance of 'std::length_error'
  what():  vector::_M_default_append`

Would be grateful for directions (if any) - i think i cannot move it further by myself.

davidgraeff, 2018-01-11:
Try to imitate any other of the supported DSOs. If it doesn't work, at least a bit, there is probably some research necessary to find determine the correct commands.

jimbeggs, 2018-02-04:
I am having similar problem with 6074.

I have examined the fwget.sh script and noticed that the 6022 drivers are not downloaded and extracted into firmware and loader.  Why is this?  Is it because that device doesn't need firmware to be downloaded?  Perhaps that would be true for the 6074 as well?

Assuming I don't have to worry about downloading firmware to the 6074, my problem would stem from a difference in messages between 6022 and 6074, so it would be good to know about firmware download so I could concentrate on reverse engineering the messaging.

alexanderustinov, 2018-02-05:
It doesn't download the firmware because hantek changed the scheme of file storage on their site - script gets 404 when tries to download files.
But there is more: if one takes files from cd extractor still could not handle new firmware data, so it seems we need to rewrite it.
For now i ended up using windows vm with installed drivers for uploading firmware.

BillFleming, 2018-07-23:
I also have the DSO-6082BE and just got the program compiled from source on linux.
```
lsusb
Bus 001 Device 016: ID 04b4:6081 Cypress Semiconductor Corp. 
```
Where should I start with trying to dump the firmware? If I can get this one working I am also interested in purchasing an even better Hantek scope to test with as well. (one of the 4 channel models possibly)

matuszpd, 2018-12-20:
I managed to extract the dso6082be-firmware (the dso6082be-loader is exactly the same as dso6022be-loader) from the windows driver if anyone interested. I tried and added the scope into the source but I'm still getting `Sending control command eX failed: Pipe error`.

alexanderustinov, 2019-01-17:
@matuszpd 
Tree has two files for dso6022be:
openhantek/res/firmware/dso6022be-firmware.hex
openhantek/res/firmware/dso6022be-loader.hex

Can you confirm that both are same for 6082be?

matuszpd, 2019-01-25:
No, as I mentioned only the loader is exactly the same, the firmware is different.

alexanderustinov, 2019-09-04:
@matuszpd
Would you please make a PR with extracted files?