OpenHantek/openhantek #150: DSO-2150 doesn't show any data

DanielO, 2018-01-11:
I have a DSO-2150 and while OpenHantek connects to it OK I don't see any data on the display (even when trigger is set to Wait/Force)

I have tried twiddling many options (eg capture buffer size, time base, etc) but no change.

davidgraeff, 2018-01-11:
Can you compile in debug mode? This way all commands send to the scope are printed out.

DanielO, 2018-01-11:
Sure.. How? :)
I couldn't see anything in the readme or docs and I am a cmake noob.

davidgraeff, 2018-01-12:
`cmake -DCMAKE_BUILD_TYPE=Debug` on the command line or use QtCreator and select Debug near the bottom in the left pane.

DanielO, 2018-01-14:
OK I get the following output..
````
[Maelstrom 14:35] ~/projects/openhantek/build >./openhantek/OpenHantek
compile shaders
compile shaders
14:35:13.104: Sending bulk command: 01 00 01 02 00 00 01 00 00 00 00 00
14:35:13.104: Sending bulk command: 07 00 00 00 00 00 00 00
14:35:13.105: Sending control command b5: 00 04 08 fd 20 40 ef 01 00 00 00 00 00 00 00 00 00
14:35:13.192: Sending control command b4: ff 00 ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00
14:35:13.240: Starting to capture
14:35:13.252: Enabling trigger
14:35:13.263: Forcing trigger
14:35:13.275: Received packet 1
0
0
````
FWIW I am at git hash eaf29e5

DanielO, 2018-01-14:
If I set trigger to Single and press the play button I get
````
14:36:26.667: Starting to capture
14:36:26.678: Enabling trigger
14:36:26.689: Forcing trigger
14:36:26.703: Received packet 4
QObject::startTimer: Timers cannot be started from another thread
QObject::startTimer: Timers cannot be started from another thread
QObject::startTimer: Timers cannot be started from another thread
0
0
````
If I select Wait/Force and press play I get
````
14:37:02.203: Starting to capture
14:37:02.215: Enabling trigger
14:37:02.226: Forcing trigger
14:37:02.238: Received packet 37
0
0
14:37:02.252: Starting to capture
14:37:02.263: Enabling trigger
14:37:02.274: Forcing trigger
14:37:02.288: Received packet 38
0
0
14:37:02.302: Starting to capture
14:37:02.314: Enabling trigger
14:37:02.326: Forcing trigger
14:37:02.338: Received packet 39
0
0
14:37:02.349: Starting to capture
14:37:02.360: Enabling trigger
14:37:02.372: Forcing trigger
14:37:02.384: Received packet 40
0
0
````
(repeatedly until I press it again)

Also FWIW if I move the trigger level it sends this command
````
14:38:33.215: Sending control command b4: ff 00 ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00
````
but it is always the same (I was expecting the command to vary slightly based on the trigger level)